fwupd/firmware-dell #1: Dell TB16: Firmware updates marked as being in testing state

paulmenzel, 2019-09-14:
The firmware updates pages [here](https://fwupd.org/lvfs/device/com.dell.mst0a52c8c7.firmware) and [here](https://fwupd.org/lvfs/device/com.dell.mstbe025b25.firmware) both say that the firmware versions are not production ready.

> Warning: This firmware is in the testing state and may not be suitable for production systems.

They also list it for version 3.10.002, which is actually installed on the docking station we have here.

superm1, 2019-09-18:
@hughsie so I think what we'll need to do to bring this to stable is come up with and document an "unbricking" method for fwupd when the device is in a bad state in case someone unplugged mid update.

BjoernDaase, 2020-08-24:
@superm1 Out of personal interest: What might brick if the dock is disconnected during an update? I somehow expected a process like: The pc fetches firmware update -> The firmware is transferred to the dock -> When the transfer is completed, the dock can update itself autonomously (by that amount of functionality docks provide these days they must have an autonomous CPU/micro-controller, no)?.

superm1, 2020-08-24:
@BjoernDaase The MST controller.  The particular model in the TB16 only has one bank, and if the firmware update is interrupted it will boot in bootloader mode (ROM with reduced featureset) and not identify properly to allow recovery.

The MST controller in the WD19 has necessary resiliency  to be safe to be unplugged "mid" update.

BjoernDaase, 2020-08-24:
@superm1 Thanks for that explanation! I guess it is same with the WD15 dock as well?

superm1, 2020-08-24:
Yes TB16 and WD15 share same MST controller flow.

andrejpodzimek, 2024-09-12:
What does _“in testing state”_ actually mean in this context? As already pointed out [under this bug](https://github.com/fwupd/fwupd/issues/1671), there is no firmware for the TB16 dock in the `lvfs-testing` remote. At least in the sense that `enable-remote lvfs-testing`, `refresh` and `update` won’t find anything.

Is there any way to install [this](https://fwupd.org/lvfs/devices/com.dell.mst0a52c8c7.firmware) or [this](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=7tj1h&oscode=w732&productcode=dell-thunderbolt-dock-tb16&lwp=rt) (other than the ones [already tried](https://github.com/fwupd/fwupd/issues/1671#issuecomment-2347246531))?