fwupd/firmware-dell #5: Precision 5530 2-in-1 support?

ArchangeGabriel, 2019-09-14:
Hi,

This is especially addressed @superm1. ;)

I’m considering to buy a Precision 5530 2-in-1. From what I read in fwupd/lvfs-website#288 and the device list, there seems to be no support for upgrading the firmware using fwupd. Will that ever change? If not, is there a manual procedure for upgrading the firmware?

Also, I’ve seen in the device list a TB firmware upgrade for the XPS 9575: https://fwupd.org/lvfs/device/com.dell.tbt8926f505.firmware

First, there is a [newer version at Dell](https://www.dell.com/support/home/fr/fr/frbsdt1/drivers/driversdetails?driverid=p1x1w&oscode=wt64a&productcode=xps-15-9575-2-in-1-laptop), second, this does not seems to apply for the Precision, which has [slightly different version numbers](https://www.dell.com/support/home/fr/fr/frbsdt1/drivers/driversdetails?driverid=ptkwv&oscode=wt64a&productcode=precision-15-5530-2-in-1-laptop). Is support for TB firmware upgrade through fwupd expected for these machines, and else is there again a manual upgrade procedure?

Thanks in advance!

ArchangeGabriel, 2019-11-25:
@superm1 Does the new “exe” workaround works in this case?

superm1, 2019-11-25:
@ArchangeGabriel yes you can work around it with that for the BIOS.

dell-client-linux, 2021-05-07:
Close the issue as Precision 5530 2-in-1 should use the exe workaround.

closed 2021-05-07 by dell-client-linux

ArchangeGabriel, 2021-05-07:
What about TB firmware update though?