fwupd/firmware-dell #6: Dell XPS 9365 firmware update missing

ausil, 2019-09-14:
For missing firmware please provide the following items:
this is a follow on to fwupd/lvfs-website#189 as a new update is out yet not available
@superm1 ping, can we please fix the process for releasing updates

* Link to firmware update at OEM website
https://www.dell.com/support/home/us/en/19/drivers/driversdetails?driverId=988JY&osCode=BIOSA&productCode=xps-13-9365-2-in-1-laptop
* Firmware release date (if available on OEM website)
11 Jan 2019
* Link to the device page on LVFS.
https://fwupd.org/lvfs/device/d636245a-c31b-4454-b9a5-9893c6e6f6cc

holzi91, 2019-09-15:
@hughsie ping, another update is available.

- Link to firmware update at OEM website
[https://www.dell.com/support/home/en/en/debsdt1/drivers/driversdetails?driverid=xc45d&oscode=wt64a&productcode=xps-13-9365-2-in-1-laptop](url)

- Release date 11 Sep 2019

- Link to device page on LVFS
[https://fwupd.org/lvfs/device/com.dell.uefid636245a.firmware](url)

dell-client-linux, 2021-05-07:
This issue should be closed as latest FW is synced to Dell website.

closed 2021-05-07 by dell-client-linux