fwupd/firmware-dell #10: Recent BIOS updates for Dell Precision 5530 not available

lgp171188, 2019-09-30:
My Dell Precision 5530 laptop is running BIOS version 0.11.2 installed using `fwupd`. There have been newer, critical releases after that - 0.12.0 released in June 2019 and 0.13.x released a few days back. However, these are not available through `fwupd`. This looks like an issue with the OEM not uploading the updates to LVFS, but since I am not a 100% sure of it, I am creating this issue. Please close it if this an OEM issue.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.10
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.10
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Precision 15 M5530  Thunderbolt Controller
  DeviceId:             3edbc5e81978be9b0aabe51ca0329fc548cb2a8a
  Guid:                 a35a21e3-7eb7-56d5-89c8-6e8b28d33ac2
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-09-30

Precision 5530 System Firmware
  DeviceId:             66352e92895dc6d018071ad4ecfae249b0b8af0d
  Guid:                 e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.11.2
  VersionLowest:        0.1.11.2
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-09-30

KXG50ZNV512G NVMe TOSHIBA 512GB
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
  Guid:                 98d34383-a939-5b5a-9806-e05e0b288b1a
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              AADA4105
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-30
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* Linux Firmware Updater        HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x100000)/File(\EFI\manjaro\fwupdx64.efi)
Boot0001* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1    HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x100000)/File(\EFI\boot\bootx64.efi)..BO
Boot0005* rEFInd Boot Manager   HD(1,GPT,22b79034-7ff8-4248-8186-235923f2c5e4,0x800,0x113000)/File(\EFI\refind\refind_x64.efi)
```

```shell
[... 322 lines cut ...]
???     ??? xnu_uuid.mod
???     ??? xnu_uuid_test.mod
???     ??? xzio.mod
???     ??? zfscrypt.mod
???     ??? zfsinfo.mod
???     ??? zfs.mod
???     ??? zstd.mod
??? initramfs-4.19-x86_64-fallback.img
??? initramfs-4.19-x86_64.img
??? intel-ucode.img
??? linux419-x86_64.kver
??? memtest86+
??? ??? memtest.bin
??? vmlinuz-4.19-x86_64

11 directories, 318 files
```

Please answer the following questions:

- Operating system and version: Manjaro latest
- How did you install fwupd? `pacman`
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-10-01:
I've sent a notification internally on this one.

superm1, 2019-10-02:
Should be available now.

closed 2019-10-02 by superm1