fwupd/firmware-dell #13: Dell Latitude 7X90 (7290/7390/7490) missing 1.13.1 BIOS

dhensen, 2020-01-02:
For missing firmware please provide the following items:

* Link to firmware update at OEM website:
<https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverid=f6n5h&oscode=wt64a&productcode=latitude-14-7490-laptop&lwp=rt>

* Firmware release date (if available on OEM website):
Version 1.13.1 release date: `2019-12-04` (YYYY-MM-DD)

* Link to the device page on LVFS:
<https://fwupd.org/lvfs/devices/a86a3f07-b7d3-4545-9238-0274fc1ba682>

:heart:

dhensen, 2020-01-03:
I am just seeing that this issue is transfered! I was not aware of this dell specific repository. I read the readme and performed the bios update manually via supplied instructions, my bios is now at `1.13.1`.

(So I no longer have a need for this, but I hope this will still benefit others)

superm1, 2020-01-06:
Thanks, forwarded to the correct folks.

superm1, 2020-01-07:
it's there now.

closed 2020-01-07 by superm1