fwupd/firmware-dell #15: Latitude 7290/7390/7490 Thunderbolt firmware version 41

stblr, 2020-01-25:
https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverid=7h18f&oscode=wt64a&productcode=latitude-14-7490-laptop

https://fwupd.org/lvfs/devices/com.dell.tbtae4a91ee.firmware

superm1, 2020-01-31:
Which system do you actually want?  Although they were bundled together for Windows, getting the CAB for Linux may be different.

stblr, 2020-02-01:
7490

The link for the 7490 on https://fwupd.org/lvfs/devices/ actually redirects to the 7290 device page so I thought they were all the same.

superm1, 2020-02-03:
You can check and see if this works for you:
https://fwupd.org/downloads/61dc47fff5030276205c58340b20ee29b2e930e9-ThunderboltFirmwareUpdateLinux_4.41.13.001.cab

If so, there are some things blocking it getting promoted right now that need to be sorted.

superm1, 2020-02-03:
It looks like that one is not available right now either, sorry.