fwupd/firmware-dell #16: Many Dell firmwares not updated to latest on LVFS

grigi, 2020-02-05:
Many Dell firmwares that used to get updated stopped getting updates in the last two months.
My system is a latitude 5490 so I list links here, but I noticed the same with 5x91.
The firmwares include security fixes, so I would prefer that they get updated.

Thanks in advance!

* Link to firmware update at OEM website: v1.12.1 (19 Dec 2019)
https://www.dell.com/support/home/za/en/zabsdt1/drivers/driversdetails?driverid=tpt7x&oscode=wt64a&productcode=latitude-14-5490-laptop

* Link to the device page on LVFS: v1.11.1 (4 Nov 2019)
https://fwupd.org/lvfs/devices/com.dell.ueficdcae5ae.firmware

superm1, 2020-02-05:
@grigi I'll ping the right people to figure out why this didn't promote to stable yet.

@pshem please file a separate issue to track this for your system.

grigi, 2020-02-06:
@superm1 Thank you, the firmwares have been updated! :+1:

closed 2020-02-06 by grigi