fwupd/firmware-dell #18: Unable to update BIOS on XPS 7390 with secure boot enabled

fredoche, 2020-04-01:
Hi,

I just got the latest dell xps 13 developper edition 7390 with the core i7 10th gen cpu -not 2in1) . I removed Ubuntu and installed fedora 31.

I cannot update its firmware with gnome-software. After clicking update, and rebooting, supposedly to run the update program, it show four lines of logs in the upper left of the screen, too fast to be read (i think it says something about a path being added), and then reboot again, this time normally and goes back to GDM and the usual user session.

```shell
$ fwupdmgr --version
fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11

```
```shell
$ fwupdmgr get-devices
XPS 7390 Thunderbolt Controller
  DeviceId:             bdf5329a388f2b37c11ab55433343b83ee6fbd6c
  Guid:                 7d538854-204d-51b2-8f9d-1fe881c70200
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              41.02
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-09-30

XPS 13 7390 System Firmware
  DeviceId:             b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  Guid:                 cbe49cca-492b-93e8-b0f4-f693501f271b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.1.0.0
  VersionLowest:        0.1.0.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-09-30
  UpdateState:          success

Micron 2200S NVMe 512GB
  DeviceId:             c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
  Guid:                 b29d3ccf-500b-5b69-9bb9-e46b4832225f
  Guid:                 d63070ee-eeed-11e8-8eb2-f2801f1b9fd1
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot|no-auto-instance-ids
  Vendor:               Micron Technology Inc
  VendorId:             NVME:0x1344
  Version:              22001010
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-09-30

```

```shell
$ efibootmgr -v
BootCurrent: 0005
Timeout: 0 seconds
BootOrder: 0005,0001,0002,0000
Boot0000* UEFI Micron 2200S NVMe 512GB 193022D9A71A 1	PciRoot(0x0)/Pci(0x1d,0x4)/Pci(0x0,0x0)/NVMe(0x1,00-A0-75-01-22-D9-A7-1A)/HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0001* ubuntu	HD(1,GPT,f4874422-1a07-42dd-a9a6-4dfd6b68a838,0x800,0x186000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux Firmware Updater	HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0005* Fedora	HD(1,GPT,9183e0e0-7491-40f7-b911-763acfd8758e,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)

```

```shell
$ efivar -l | grep fw
** nothing **
```

```shell
$ tree /boot
/boot
├── config-5.3.0-0.rc6.git0.1.fc31.x86_64
├── config-5.3.1-300.fc31.x86_64
├── efi
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTIA32.EFI
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fbia32.efi
│   │   │   └── fbx64.efi
│   │   └── fedora
│   │       ├── BOOTIA32.CSV
│   │       ├── BOOTX64.CSV
│   │       ├── fonts
│   │       │   └── unicode.pf2
│   │       ├── fw
│   │       ├── fwupdx64.efi
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       └── shimx64-fedora.efi
│   ├── mach_kernel
│   └── System
│       └── Library
│           └── CoreServices
│               └── SystemVersion.plist
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-01d1fd354a0e48b7ac999d422c01d3a1.img
├── initramfs-5.3.0-0.rc6.git0.1.fc31.x86_64.img
├── initramfs-5.3.1-300.fc31.x86_64.img
├── loader
│   └── entries
│       ├── 01d1fd354a0e48b7ac999d422c01d3a1-0-rescue.conf
│       ├── 01d1fd354a0e48b7ac999d422c01d3a1-5.3.0-0.rc6.git0.1.fc31.x86_64.conf
│       └── 01d1fd354a0e48b7ac999d422c01d3a1-5.3.1-300.fc31.x86_64.conf
├── lost+found
├── memtest86+-5.01
├── System.map-5.3.0-0.rc6.git0.1.fc31.x86_64
├── System.map-5.3.1-300.fc31.x86_64
├── vmlinuz-0-rescue-01d1fd354a0e48b7ac999d422c01d3a1
├── vmlinuz-5.3.0-0.rc6.git0.1.fc31.x86_64
└── vmlinuz-5.3.1-300.fc31.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 31 beta
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): its not installed, but fwupdmgr is
- Have you tried rebooting? yes 6 or 7 times
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)?
 yes
 mokutil --sb-state
SecureBoot enabled

Thanks for your help

fredoche, 2020-04-08:
Just had the issue by trying to upgrade to fw 1.5.1 today.
Cheers.

demkantor, 2020-05-01:
same, ubuntu 19.10 cant update to 1.5.1
going to try some of the tips reported here and reply back if any work

mikran, 2020-05-13:
same, ubuntu 20.04

> Upgrade available for System Firmware from 1.4.0 to 1.5.1
XPS 13 7390

demkantor, 2020-05-13:
> same, ubuntu 20.04
> 
> > Upgrade available for System Firmware from 1.4.0 to 1.5.1
> > XPS 13 7390

Sorry forgot to get back, so nothing in this thread worked for me so I downloaded the update from Dell and manually installed in BIOS, after that it all worked... Until I rebooted and it refused to boot, so I made a USB stick with the 20.04 image on, ran boot repair to fix grub, then updated to 20.04. not a clean install but update so all my files are intact. 
Now everything is working as expected and on proper bios firmware,

fredoche, 2020-11-19:
Today's firmware update mentions
  • Fixed the issue where the Linux Vendor Firmware Service (LVFS) does not work when Secure Boot is enabled in the BIOS setup menu.
Cant wait to try this out. If anyone from Dell is looking, thanks for keeping "old" models up to date. I'll let you know if it works.

superm1, 2020-11-19:
>If anyone from Dell is looking, thanks for keeping "old" models up to date.
Glad it finally landed.

@fredoche As the fix is to executable code in the BIOS itself, upgrading to it will still fail, but the next BIOS after this (or reinstalling this one) will work.

closed 2020-11-19 by superm1

fredoche, 2020-11-19:
Seems like there is another issueUpgrade available for System Firmware from 1.6.0 to 1.7.0
XPS 13 7390 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Téléchargement…          [***************************************]
Erreur lors de la lecture de données du connecteur TLS: Erreur de décodage du paquet TLS reçu.

superm1, 2020-11-19:
Can you open a separate issue for that back at fwupd project?

demkantor, 2020-11-19:
I haven't a chance to test out the new firmware so thanks for pointing it out! I'll let you all know if it installs perfect on its own (like it did last time for me) or if I need to do it manually again

fredoche, 2020-11-19:
I created https://github.com/fwupd/fwupd/issues/2607

superm1, 2020-11-19:
> I haven't a chance to test out the new firmware so thanks for pointing it out! I'll let you all know if it installs perfect on its own (like it did last time for me) or if I need to do it manually again

Keep in mind it's going to still fail with current firmware + SB on.  Installing _next_ firmware upgrade or re-installing this one will be fixed.