fwupd/firmware-dell #19: Dell XPS 7390 2-in-1: TPM PCR0 differs from reconstruction

mrMoe, 2020-05-13:
**Describe the bug**
```shell
$ fwupdmgr get-devices
```
creates 
```shell
 Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
```
for my System Firmware

**Steps to Reproduce**
> fwupdmgr get-devices
as root

**Expected behavior**
successful parsing of the TPM PCR0

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version                                                                                                                                                                                                                                                 
client version: 1.4.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
> pacman -S fwupd

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices                                                                                                                                                                                                                            
XPS 13 7390 2-in-1
│
├─Thunderbolt Controller:
│     Device ID:           ec6b4dfd3088670ee6487f23f579f7607c91aa5b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     77.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Thunderbolt Controller:
│     Device ID:           03640de408f8cd76c3371fd43a80837e3b743ff5
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     77.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─WD19TB:
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.00
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       22Z0N03/3104265116064743
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
[... 128 lines cut ...]
└─Touchpad:
      Device ID:           08fadf5c169ee39d8c8f08f264108ea3c8aeee30
      Current version:     1.4.2888190
      Bootloader Version:  54.0.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               a9c1b42d-a78c-5068-9b5b-af7ee1d65c88 ← HIDRAW\VEN_06CB&DEV_CD7A&REV_00
                           58d4ca98-ced2-549c-b505-c12d49eb400b ← HIDRAW\VEN_06CB&DEV_CD7A
                           9073e287-0769-5fd5-9342-8df98eefbb6c ← SYNAPTICS_RMI\TM3450-001
                           e5b92a29-f5df-5c45-b280-c5bb2dc127e1 ← SYNAPTICS_RMI\TM3450
      Device Flags:        • Internal
```

**Additional questions**
- Operating system and version:
Arch Linux
```shell
uname -a                                                                                                                                                                                                                                                           
Linux rtfm 5.6.11-arch1-1 fwupd/fwupd#1 SMP PREEMPT Wed, 06 May 2020 17:32:37 +0000 x86_64 GNU/Linux
```
- Have you tried rebooting?
yes; multiple times
- Is this a regression?
unknown

[fwupdtpmevlog.txt](https://github.com/fwupd/fwupd/files/4604403/fwupdtpmevlog.txt)

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

mrMoe, 2020-06-18:
I just got the arch update to version 1.4.4 and am sadly to say the error still persists

```shell
$ fwupdmgr --version                                                                                       
client version: 1.4.4
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.4
```
```shell
$ fwupdmgr get-devices                                                                                     
XPS 13 7390 2-in-1
│
├─Thunderbolt Controller:
│     Device ID:           afcc9427ef083d0bb017d0010c530014d7c99aaf
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     77.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─KBG40ZPZ1T02 NVMe KIOXIA 1024GB:
│     Device ID:           ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:             NVM Express Solid State Drive
│     Current version:     10400104
│     Vendor:              Intel Corporation (NVME:0x1E0F)
│     Serial Number:       Y9C100ZENTRL
│     GUIDs:               fbef9958-2179-5fb1-9e91-31c22cc29b2d ← STORAGE-DELL-107871
│                          b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66560
│     Minimum Version:     66560
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                20448dae-1dd2-5653-a2b3-c81b77007d2f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─Touchpad:
│     Device ID:           08fadf5c169ee39d8c8f08f264108ea3c8aeee30
│     Current version:     1.4.2888190
│     Bootloader Version:  54.0.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               a9c1b42d-a78c-5068-9b5b-af7ee1d65c88 ← HIDRAW\VEN_06CB&DEV_CD7A&REV_00
│                          58d4ca98-ced2-549c-b505-c12d49eb400b ← HIDRAW\VEN_06CB&DEV_CD7A
│                          9073e287-0769-5fd5-9342-8df98eefbb6c ← SYNAPTICS_RMI\TM3450-001
│                          e5b92a29-f5df-5c45-b280-c5bb2dc127e1 ← SYNAPTICS_RMI\TM3450
│     Device Flags:        • Internal device
│                          • Updatable
│   
└─USB Keyboard:
      Device ID:           6e840ff67ccd6980286a0c4b83f4edc7df3328cb
      Summary:             USB PD
      Current version:     149.1.4.3
      Vendor:              VIA Labs, Inc. (USB:0x2109)
      Serial Number:       0000000000000001
      GUIDs:               9b630f16-62b2-57a2-a4d3-36df60ac08ad ← USB\VID_2109&PID_D101&REV_0301
                           56a509c6-b4c3-5cf3-bf9d-8c98bc8562dc ← USB\VID_2109&PID_D101
                           446d3992-2a03-5143-a99c-6189ec313d86 ← USB\VID_2109&PID_D101&DEV_VL102
      Device Flags:        • Updatable
                           • Cryptographic hash verification is available
```

superm1, 2020-06-18:
Unfortunately it's not in 1.4.4, it is a commit not yet tagged. So you would need to hand compile or wait for 1.4.5 to check it.

didierm, 2020-06-20:
COPR version fwupd-1.5.0-0.591.20200617git (https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/build/1474841/) does not seem to fix this issue : 
```
# fwupdmgr get-devices --show-all-devices          
XPS 13 7390 2-in-1
...
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66305
│     Minimum Version:     66305
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
...
```

superm1, 2020-06-21:
Thanks for confirming; then that means there is a secondary problem to that one reported.

Can you please with that build share `fwupdtpmevlog` output? and also contrast it to tpm2_readpcrs / tpm2_listpcrs (whichever command is appropriate on your system)?

The version string is actually intended behavior until an update is made available on LVFS.  You can update from the EXE file on support.dell.com in the F12 POST menu, or by using the stuff mentioned in README here: https://github.com/fwupd/firmware-dell

didierm, 2020-06-21:
Requested outputs attached.
[tpm2_pcrread.out.gz](https://github.com/fwupd/firmware-dell/files/4809341/tpm2_pcrread.out.gz)
[fwupdtpmevlog.out.gz](https://github.com/fwupd/firmware-dell/files/4809342/fwupdtpmevlog.out.gz)

superm1, 2020-06-21:
Thanks.  It looks like the others are re-calculating right, it's just PCR0 problem right now.  Can you please do the following:
1. Upgrade to latest firmware.
2. Check if this is happening on every single boot?  Or is it only on a cold boot some times, or only on a warm boot?  Just a handful of reboots on warm and a handful of cold boots should be sufficient ot see if it happens every time.

didierm, 2020-06-21:
In the meantime, I had already upgraded tot the latest BIOS FW v1.4.0.
After a warm reboot (Fedora 32, kernel 5.7.4) and subsequent suspend/resume test cycle : 
```
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66560
│     Minimum Version:     66560
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update State:        success
│     Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUIDs:               20448dae-1dd2-5653-a2b3-c81b77007d2f
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                          1d203f3b-1409-598e-92d6-2633efe29caa ← UEFI\RES_{20448DAE-1DD2-5653-A2B3-C81B77007D2F}
```
As I am using this system in production use (with some tens of windows concurrently open, divided over 8 workspaces, and uptimes of 10-30 days), restoring my desktop setup after reboot is not too pleasant, and takes some time.

OK if I test this after my next major kernel upgrade + subsequent reboot(s) ?

didierm, 2020-07-25:
Past week, I upgraded to kernel-5.7.9-200.fc32.x86_64 (from 5.7.4), and the required kernel reboot, the availability of a new Dell FW 1.5.0 and a new fwupd version (fwupd-1.5.0-0.603.20200706git) provided an excellent occasion to reassess fwupd.

1. Installing fwupd-1.5.0-0.603.20200706git (not rebooted yet, uptime 22 days, still with kernel-5.7.4 and Dell FW 1.4.0) leads to a chain of  anomalies : 

* **pesign** dependency error : 
```
# dnf install https://download.copr.fedorainfracloud.org/results/rhughes/fwupd/fedora-32-x86_64/01518729-fwupd/fwupd-1.5.0-0.603.20200706git.fc32.x86_64.rpm

"Failed to try-restart pesign.service: Unit pesign.service not found."
```
Reference : https://bugzilla.redhat.com/show_bug.cgi?id=1776680

* **efivarfs** error : 

```
# fwupdmgr get-devices --show-all-devices
...
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66560
│     Minimum Version:     66560
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        /sys/firmware/efi/efivars was not mounted
│     GUIDs:               20448dae-1dd2-5653-a2b3-c81b77007d2f
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                          1d203f3b-1409-598e-92d6-2633efe29caa ← UEFI\RES_{20448DAE-1DD2-5653-A2B3-C81B77007D2F}
│     Device Flags:        • Internal device
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
...

# mount | grep efivars
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,relatime)
# mount -t efivarfs efivarfs /sys/firmware/efi/efivars
-> same error
```

* **firmware update** error : 

```
# /usr/share/fwupd/install_dell_bios_exe.py /tmp/XPS_13_7390_2-in-1_1.5.0.exe
Installing to System Firmware
Wrote capsule /tmp/tmpldfc8hn7/firmware.bin
GUID:      20448dae-1dd2-5653-a2b3-c81b77007d2f
HdrSz:     0x1000
Flags:     0x70000
PayloadSz: 0xd62420
Generated CAB file /tmp/tmpldfc8hn7/firmware.cab
[                                                  ] 0% idle
FwupdError: Device System Firmware [fb5523b663445df99fe178d162a221546e1974af] does not currently allow updates (10)
```

2. Downgrading fwupd fixes the firmware upgrade issue : 

```
# dnf downgrade fwupd
# fwupdmgr --version
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4

# fwupdmgr get-devices --show-all-devices
...
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66560
│     Minimum Version:     66560
[... 64 lines cut ...]

# fwupdmgr get-devices --show-all-devices
...
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     66816
│     Minimum Version:     66816
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update State:        success
│     GUIDs:               20448dae-1dd2-5653-a2b3-c81b77007d2f
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                          1d203f3b-1409-598e-92d6-2633efe29caa ← UEFI\RES_{20448DAE-1DD2-5653-A2B3-C81B77007D2F}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
...
```

5. Summary : 
 * upgrading to fwupd 1.5.0-0.603.20200706git causes lots of issues all over the place ;
 * downgrading to 1.4.4 allows to upgrade the Dell FW ;
 * rebooting and re-upgrading to 1.5.0-0.603.20200706git appears to fix the TPM PCR0 issue.

hughsie, 2020-07-26:
> upgrading to fwupd 1.5.0-0.603.20200706git

This won't work if you have secure boot turned on as the binary is not signed.

> 20200706git appears to fix the TPM PCR0 issue

As expected.

didierm, 2020-07-26:
> > upgrading to fwupd 1.5.0-0.603.20200706git
> 
> This won't work if you have secure boot turned on as the binary is not signed.

Is this a general remark, or specific to my report ?
_(FYI, IIRC, secure boot was/is turned off on my system)_

superm1, 2020-07-27:
General remark, only official distro builds typically get signed.

superm1, 2020-07-27:
I'll close this issue since it's indeed fixed in master, should be available in 1.5.0+ or 1.4.5+ when tagged.

closed 2020-07-27 by superm1

mrMoe, 2020-08-16:
I can confirm it is fixed in 1.4.5 on arch linux