fwupd/firmware-dell #20: Dell XPS 13 9300 - TPM PCR0 differs from reconstruction

robotmaxtron, 2020-05-13:
**Describe the bug**
fwupdmgr get-devices shows an update error. 

> Update Error: TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

**Steps to Reproduce**
Fresh installation of Ubuntu 20.04 on Dell XPS 13 9300

**Expected behavior**
No errors

**fwupd version information**
Please provide the version of the daemon and client.
```shell
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Installed as part of the normal Ubuntu installer

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
XPS 13 9300
│
├─Thunderbolt Controller:
│     Device ID:           616618c1a56eae6fe6cb9715359364e08110dcb5
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     75.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Thunderbolt Controller:
│     Device ID:           e841ee543b2f10cc90292f884df14d3882bb32b0
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     75.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─BC501 NVMe SK hynix 256GB:
│     Device ID:           ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:             NVM Express Solid State Drive
│     Current version:     80002C00
│     Vendor:              SK hynix (NVME:0x1C5C)
│     GUIDs:               b637d847-48b0-519f-a4f1-ed2b8535d300
│                          da85e71f-d425-e811-b467-0ed5f89f718b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Integrated Webcam HD:
│     Device ID:           4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:     91.58
│     Vendor:              CN0PW36V8LG0099B00DPX02 (USB:0x0BDA)
│     GUIDs:               bba32a58-5e20-5d64-bc15-287159bdaf46
│                          14e94ffe-f85a-5645-9af7-705464a66c8c
│     Device Flags:        • Updatable
│   
├─Iris Plus Graphics G7:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
[... 361 lines cut ...]
├── initrd.img -> initrd.img-5.4.0-21-generic
├── initrd.img-5.4.0-21-generic
├── initrd.img.old -> initrd.img-5.4.0-21-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.4.0-1002-oem
├── System.map-5.4.0-21-generic
├── vmlinuz -> vmlinuz-5.4.0-21-generic
├── vmlinuz-5.4.0-21-generic
└── vmlinuz.old -> vmlinuz-5.4.0-21-generic

5 directories, 295 files
```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? Yes
- Is this a regression? Unknown, I can't recall if it always reported the error.
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes

I've attached output from /usr/bin/fwupdtpmevlog as suggested.
[fwupdtpmevlog.txt](https://github.com/fwupd/fwupd/files/4459316/fwupdtpmevlog.txt)

smithsos, 2020-05-30:
Same issue on my install. Was present in 1.0.7 and persists after update to 1.0.10.

XPS 13 9300
│
├─Thunderbolt Controller:
│     Device ID:           44a608582e1a34b95c4763f982a3424531e8a2db
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     80.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Thunderbolt Controller:
│     Device ID:           3ff580cb8f0b9f3e462f70a049ccdf3ee6d4f578
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     80.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Integrated Webcam HD:
│     Device ID:           4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:     96.28
│     Vendor:              CN0PW36V8LG00032A1WKA00 (USB:0x0BDA)
│     GUIDs:               d784b200-4ead-55f2-8277-15e1618cbe7b
│                          9a1becb6-4d9f-5855-99d0-145b1700da10
│                          d35eb209-22f9-59a5-a918-33d7307edab9
│     Device Flags:        • Updatable
│   
├─PC611 NVMe SK hynix 512GB:
│     Device ID:           ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:             NVM Express Solid State Drive
│     Current version:     11000111
│     Vendor:              SK hynix (NVME:0x1C5C)
│     GUIDs:               919c0a0c-88f2-518c-be84-39f0c34c5104
│                          89f21a68-b326-11e9-a2a3-2a2ae2dbcce4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           f2f44dcc012cdad82b823131f5c85ebfb74d782e
│     Current version:     1.0.11
│     Minimum Version:     1.0.11
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                e8292593-e66e-4878-b051-f152535ab130
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│     Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:             Platform TPM device
│     Current version:     0.74.0.8
│     Vendor:              Dell Inc. (PCI:0x1028)
│     Update Error:        Updating disabled due to TPM ownership
│     GUIDs:               02a963b2-82d6-5ab7-bd72-bcd6c2113772
│                          ff71992e-52f7-5eea-94ef-883e56e034c6
│                          73730635-f6c2-53da-9df2-948bb5ac1022
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
└─Touchpad:
      Device ID:           43284a5278d905c164569e3a3edd06c6c4aa613b
      Current version:     1.3.3040414
      Bootloader Version:  54.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               76018764-887c-558b-a6c0-ae31e4522d98
                           f6982c83-dec9-5961-a250-384441c12205
                           d69eed5b-0b5e-53c7-ac04-8e7a672f0365
                           913a517f-55c6-5f1d-b018-405af36fcbc0
      Device Flags:        • Internal device
                           • Updatable`
`

[fwupdtpmevlog.log](https://github.com/fwupd/firmware-dell/files/4705459/fwupdtpmevlog.log)

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

superm1, 2021-05-27:
Yeah you need 1.3.12 in the 1_3_X series.

closed 2021-05-27 by superm1