fwupd/firmware-dell #21: Precision 7730 PCR0 reconstruction fails

toadzky, 2020-05-14:
Model: Dell Precision 7730

As requested, here's the output from the log:

[fwupdtpmevlog.log](https://github.com/fwupd/fwupd/files/4552798/fwupdtpmevlog.log)
```
PCR:                     BIOS (0)
Type:                    0x7
Description:             EV_S_CRTM_CONTENTS
ChecksumSha1:            a6f807d160e7f182f26513f2e50006817abe500a085d8812f44ee839d4433451
BlobStr:                 Boot Guard Measured S-CRTM.

PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155
BlobStr:                 ..kT..U@..N....:

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            9cffed427db3cc30efb0ea1637069ef7b7256ff722474339097807888d475adb
BlobStr:                 ..........j.....

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000009
Description:             EV_EFI_HANDOFF_TABLES
ChecksumSha1:            0f57271e82d06cab06e58b458ad63bb406d375525694b1ee692fafb341bd7431
BlobStr:                 ...........? boD.....|....UQ....

PCR:                     BIOS Configuration (1)
Type:                    0x80000009
Description:             EV_EFI_HANDOFF_TABLES
ChecksumSha1:            9c844a8fe22e6309ce1b358b0066a1c9cf0608b563b291563c6f388c3a3149ff
BlobStr:                 ...........? boD.....|...V.Q....

PCR:                     BIOS Configuration (1)
Type:                    0xa
Description:             EV_PLATFORM_CONFIG_FLAGS
ChecksumSha1:            f58e6c016150ccdbdec3d3c3fc54a03c03460256eac65583a54fb2bdf687d815
BlobStr:                 ..........0........."....#...................B....J..............a....`...._....D$..................|.........x....7....)....#...."....$....b..............2.........

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            faa9c96bd54ba27f521338c1629ec6e7e329555eb806d2ba66b48172575cc89d
BlobStr:                 a.............+.................B.o.o.t.O.r.d.e.r.....

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            4a0702a7accea5230bbdc9b85834f647c37cd6db317ff3aa03d3ecd44e2a564d
BlobStr:                 a.............+.........v.......B.o.o.t.0.0.0.0.....b.u.b.u.n.t.u.....*.........................................4.\.E.F.I.\.u.b.u.n.t.u.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            f3b9709de4f6b259e3cbdd2f8a17d340b61f9cc39cf055765fcfa93cb096eebf
BlobStr:                 a.............+.................B.o.o.t.0.0.0.1.......U.E.F.I.:. .P.C.4.0.0. .N.V.M.e. .S.K. .h.y.n.i.x. .5.1.2.G.B.,. .P.a.r.t.i.t.i.o.n. .2.....*.........................................0.\.E.F.I.\.b.o.o.t.\.b.o.o.t.x.6.4...e.f.i.........8..Gd-.;.A..MQ..L.M.J.8.7.N.4.3.0.3.1.0.1.0.7.A.2.N.........BO

[... 474 lines cut ...]
ChecksumSha1:            668b315729c39ad5f0a0b59f870b607776edeb425481b70a1de01b0a19c12e2d
BlobStr:                 (hd0,msdos5)/boot/grub/fonts/unicode.pf2.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            b32a0876f9b7c8cc4769e890dc2ebd15d0beb65d710f2896e5f852c6d42cddb5
BlobStr:                 /boot/vmlinuz-5.4.0-28-generic.

PCR:                     Static operating system (9)
Type:                    0xd
ChecksumSha1:            da3462454052012305f39e0af41c910f94c3394bb1453bca19108195d18be5e8
BlobStr:                 /boot/initrd.img-5.4.0-28-generic.

PCRs:
  0:                     abb86fc17d7390bdefc60491193022d0db0333a0
  1:                     8f7fd77c2c7df0aa79b550a16b62ed126cb0d8cf
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     8b3422f26c1b96b0d9ba01807f434ea69525b255
  5:                     3a4285c323c5af121f270900e88ffeefb2a34233
  6:                     779c58e337edbdebc708901c952a8c6bfd3a00ea
  7:                     205cf7255dc8264c50f05035c061b9243c0d4beb
  8:                     0eb2644a1f555c7997a2a2c3a7033cad3a790e26
  9:                     fcbcca94fbba12a55c0cb1420833978f3ff3aadb
```

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

superm1, 2021-02-01:
We're fairly certain this is fixed in current fwupd.  If you can still reproduce with a current fwupd release, please open a new issue and link back to this one.

closed 2021-02-01 by superm1