fwupd/firmware-dell #22: Latest Latitude 5491 firmware is missing

sitnik, 2020-05-18:
Dell site shows that BIOS version 1.12.0 is available and "urgent". fwupd.org lists 1.11.1 as the latest.

I have a boot issue with version 1.11.1 and would like to check if it is fixed in the latest version.

For missing firmware please provide the following items:

* Link to firmware update at OEM website
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=6pt28&oscode=biosa&productcode=latitude-14-5491-laptop&lwp=rt

* Firmware release date (if available on OEM website)
 12 May 2020 

* Link to the device page on LVFS.
https://fwupd.org/lvfs/devices/com.dell.uefi479d7d09.firmware

superm1, 2020-05-18:
Thanks, I've sent a notification internally to find out why this one isn't yet promoted.  If it's time sensitive for your side, you can follow the instructions in README.md in this repo, otherwise I _expect_ this should be promoted soon.

sitnik, 2020-05-19:
Thanks, I'll wait till a new firmware arrive in fwupd and report back here.

superm1, 2020-05-22:
it's there now

closed 2020-05-22 by superm1

sitnik, 2020-05-22:
Installed it just fine with fwupd.

Version 1.12.0 fixed boot issues describe by other people in these threads on the official Dell community forums:
1.  https://www.dell.com/community/Latitude/Latitude-5591-Won-t-Boot-Unless-Power-Cord-Connected/td-p/7494842
2. https://www.dell.com/community/Latitude/Linux-does-not-boot-on-Latitude-5591-when-on-battery/td-p/7494242

I hope wouldn't return if Intel releases a new microcode for CPU in this laptop.

superm1, 2020-05-26:
Huh... That's interesting, thanks for sharing!  This sounds like it might be: https://bugs.launchpad.net/ubuntu/+source/intel-microcode/+bug/1862751