fwupd/firmware-dell #23: Precision 7740 PCR0 reconstruction fails

hpvb, 2020-05-21:
Probably related to #21 but here goes:

I have no idea if this used to work with older firmware versions. I only found out today this is a problem on this machine.

```
# fwupdmgr get-devices
Precision 7740
│
├─Thunderbolt Controller:
│     Device ID:           77954416d56a4d84c7830c713a347c3b74d5fa7d
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     41.01
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               49c8274e-bc6a-5707-a197-ca0c9a35bda0 ← THUNDERBOLT\VEN_00D4&DEV_0927&REV_00
│                          3f759afe-e19e-5d26-bb64-0aad84c82ac3 ← THUNDERBOLT\VEN_00D4&DEV_0927
│                          81db9688-b112-5e24-9f92-e96f05770934 ← TBT-00d40927-native
│                          1707b992-6eb5-5913-8601-6ac003d0f6c1 ← TBT-00d40927-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─KXG50PNV2T04 NVMe KIOXIA 2048GB:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     AFDA4105
│     Vendor:              Toshiba Corporation (NVME:0x1179)
│     Serial Number:       X9OA21YFKKTL
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab ← NVME\VEN_1179&DEV_0116&REV_00
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9 ← NVME\VEN_1179&DEV_0116
│                          82cb5521-56c1-59fd-92ff-6efc210afb7e ← KXG50PNV2T04 NVMe KIOXIA 2048GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           cac9de2369fa3c52f34448afb98d95c9d18cc6d4
│     Current version:     1.8.2
│     Minimum Version:     1.8.2
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Message:      TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUIDs:               74992bad-d49d-4f82-bb77-bbe865bea34e
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│                          c2b8844c-c82e-5ce5-8dda-753d175d8e31 ← UEFI\RES_{74992BAD-D49D-4F82-BB77-BBE865BEA34E}
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─TPM 2.0:
      Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:             Platform TPM device
      Current version:     7.2.1.0
      Vendor:              Dell Inc. (PCI:0x1028)
      GUIDs:               ac28967e-88af-5d62-b69e-0671a22a0a7e ← 0927-2.0
                           ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                           7d65b10b-bb24-552d-ade5-590b3b278188 ← DELL-TPM-2.0-NTC-NPCT
                           6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d ← DELL-TPM-2.0-NTC-NPCT75x
                           fe462d4a-e48f-5069-9172-47330fc5e838 ← DELL-TPM-2.0-NTC-NPCT75xrls
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
```

[... 298 lines cut ...]
  1 : 0xC04CB07F96FFEF69D88B100145770A01ADFB97FDB0FC887CE62F86041A60DA3E
  2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0x16A16B50AEE0856E093AAF8C10E313C2C764EEA09009D97600B33B39251C12B5
  5 : 0x411C47151694B2A235CE245B482BE2399157FEDE60F116219A016F0EE37CC57C
  6 : 0x88047A2BF5FA014719FB3E7136E7B847F1E86DF4AF007553F089962EA85ED88A
  7 : 0xB523EC60C0748F15C1F39A60CF582B8BD4A746FF644517984820E771F2475B86
  8 : 0x0000000000000000000000000000000000000000000000000000000000000000
  9 : 0x0000000000000000000000000000000000000000000000000000000000000000
  10: 0xA04FB418325A1773E35B68EB73AF7646A6D80F31F2F82BF4110B6F256F31FE13
  11: 0x0000000000000000000000000000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000000000000000000000000000
  14: 0x0000000000000000000000000000000000000000000000000000000000000000
  15: 0x0000000000000000000000000000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000000000000000000000000000
sha384:
```

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

hpvb, 2020-10-27:
Tested on 1.5.0 and I get this now:
```
WARNING: UEFI ESP partition not detected or configured
Host Security ID: HSI:2 (v1.5.0)

HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override:                 Locked
✔ CSME v0:12.0.68.1606:          Valid
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel DCI debugger:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✘ Intel CET Enabled:             Not supported
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```
So it looks OK to me!

closed 2020-10-27 by hpvb