fwupd/firmware-dell #24: Dell XPS 15 7590, System Firmware "TPM PCR0 differs from reconstruction"

Tyilo, 2020-05-26:
**Describe the bug**
I recently updated the system firmware from 1.6.0 to 1.7.0 on my Dell XPS 15 7590 running Arch Linux.
I'm not sure if this error was shown before upgrading to 1.7.0.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
I installed it using `pacman -Syu fwupd`.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
XPS 15 7590
│
├─Thunderbolt Controller:
│     Device ID:           d6311a736bc791be78bf4efb8c57d07c2e786a58
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               dc6ebd31-1fe5-5230-8090-dd0b06da4f87
│                          dd244205-fa73-594e-b5ce-3aab372825e0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Dell WD15:
│ │   Device ID:           ed9027c86c336ba5d3312f83c06f78b2ea6ef915
│ │   Summary:             A USB type-C docking station
│ │   Current version:     0.0.0
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:         29bca9f02e33e23330ae801ae7d6c22eb2b758e0
│ │     Summary:           A USB type-C docking station
│ │     Current version:   0.3.18
│ │     Vendor:            Dell Inc. (PCI:0x1028)
│ │     GUID:              fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │   
│ └─Dell WD15 Port Controller 1:
│       Device ID:         27c32815e5fed8f507c0210195bca394b635780f
│       Summary:           A USB type-C docking station
│       Current version:   1.1.8
│       Vendor:            Dell Inc. (PCI:0x1028)
│       GUID:              8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│     
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
[... 298 lines cut ...]
BlobStr:                 Security Level is Downgraded to 0

PCR:                     Static operating system (8)
Type:                    0xd
ChecksumSha1:            3d59747e31ffcafd70a4d1665fb9926a9a87c3251598110bc0b5c1aea110414c
BlobStr:                 i.n.i.t.r.d.=.\.i.n.t.e.l.-.u.c.o.d.e...i.m.g. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x...i.m.g. .c.r.y.p.t.d.e.v.i.c.e.=.U.U.I.D.=.9.d.3.9.a.6.5.6.-.9.8.6.2.-.4.2.3.4.-.8.7.7.6.-.2.7.0.f.0.9.c.0.e.3.9.a.:.c.r.y.p.t.r.o.o.t. .r.o.o.t.=./.d.e.v./.m.a.p.p.e.r./.c.r.y.p.t.r.o.o.t. .r.w. .m.e.m._.s.l.e.e.p._.d.e.f.a.u.l.t.=.d.e.e.p..

PCRs:
  0:                     8922847b65af87f201866ee4b6a56661c76dcae4
  1:                     a461455faf01ff02dd59690744b4af82bb061013
  2:                     3a4285c323c5af121f270900e88ffeefb2a34233
  3:                     3a4285c323c5af121f270900e88ffeefb2a34233
  4:                     b75ec256500fb2cc28d45a6cf437894c49c4e388
  5:                     a638215a7b6ab60edebabf0debf4cbd9913eee57
  6:                     fcc302ec39326e4a490968b004958116079f67aa
  7:                     57b57b810bb3b247876071db14736545036de48a
  8:                     1b1033a006a3803e61157be1c69390672ce31ddd
```

**Additional questions**
- Operating system and version: Arch Linux, fully updated
- Have you tried rebooting?
Yes
- Is this a regression?
I don't know.

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

Tyilo, 2020-12-11:
This seems to have been fixed.

closed 2020-12-11 by Tyilo