fwupd/firmware-dell #25: TPM PCR0 differs from reconstruction: Execution log Dell XPS 13" 2-in-1

sejo, 2020-05-29:
**Describe the bug**
https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

**Steps to Reproduce**
fwupdmgr get-devices

**Expected behavior**
No such message available

**fwupd version information**
root@octomore:~# fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt (ubuntu 20.04)

**fwupd device information**
root@octomore:~#  fwupdmgr get-devices --show-all-devices
XPS 13 7390 2-in-1
│
├─Thunderbolt Controller:
│     Device ID:           c46666f10d592aa08fe2fe193c27528ed9fbabee
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     77.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Thunderbolt Controller:
│     Device ID:           1e5b1390cc8cf9b9f33deccf7aebc33f9684e8de
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     77.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
├─Iris Plus Graphics G7:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     07
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               c213be4c-faf8-5fe6-9430-b458722b2656 ← PCI\VEN_8086&DEV_8A52&REV_07
│                          43ede583-c357-55e5-854d-f6f15dab900d ← PCI\VEN_8086&DEV_8A52
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─KBG40ZPZ1T02 NVMe KIOXIA 1024GB:
│     Device ID:           ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:             NVM Express Solid State Drive
│     Current version:     10400104
│     Vendor:              Intel Corporation (NVME:0x1E0F)
│     Serial Number:       Y9C100FUNTRL
│     GUIDs:               fbef9958-2179-5fb1-9e91-31c22cc29b2d ← STORAGE-DELL-107871
│                          b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           fb5523b663445df99fe178d162a221546e1974af
│     Current version:     1.3.1
│     Minimum Version:     1.3.1
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
│     GUID:                20448dae-1dd2-5653-a2b3-c81b77007d2f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     74.8.17568.5511
│ │   Vendor:              ST Microelectronics (TPM:STM)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│ │                        84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│ │                        bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│ │                        70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│ │                        06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            ST Microelectronics (TPM:STM)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:      • Internal device
│     
└─Touchpad:
      Device ID:           08fadf5c169ee39d8c8f08f264108ea3c8aeee30
      Current version:     1.4.2888190
      Bootloader Version:  54.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               a9c1b42d-a78c-5068-9b5b-af7ee1d65c88 ← HIDRAW\VEN_06CB&DEV_CD7A&REV_00
                           58d4ca98-ced2-549c-b505-c12d49eb400b ← HIDRAW\VEN_06CB&DEV_CD7A
                           9073e287-0769-5fd5-9342-8df98eefbb6c ← SYNAPTICS_RMI\TM3450-001
                           e5b92a29-f5df-5c45-b280-c5bb2dc127e1 ← SYNAPTICS_RMI\TM3450
      Device Flags:        • Internal device
                           • Updatable
   

**Additional questions**
- Operating system and version: Ubuntu 20.04 | Linux octomore 5.4.0-31-generic fwupd/fwupd#35-Ubuntu SMP Thu May 7 20:20:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
- Have you tried rebooting? Not applicable
- Is this a regression? No

superm1, 2020-05-29:
This looks like its should likely be a duplicate of #19.

superm1, 2020-06-15:
A [bug](https://github.com/fwupd/fwupd/pull/2183) with PCR0 reconstruction was identified in fwupd code.  This bug has been fixed in the stable branches for all applicable releases: [1_3_X,](https://github.com/fwupd/fwupd/commit/57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e) [1_4_X ](https://github.com/fwupd/fwupd/commit/67b2dca0752d3d870bc7a76c8f0945395d92e379) and [master](https://github.com/fwupd/fwupd/commit/5b63015c19b4bb3c2e0b38326e10277931a96002).

Can you please upgrade to a version with the fix, and confirm if this behavior still happens?

mrMoe, 2020-06-15:
Arch linux current version of fwupd is 1.4.2 which still shows the same message. I'll recheck as soon as the package got updated.

superm1, 2020-06-15:
The fix just released today, not included in any release yet. If you can check from source yourself just need to pull in the one line patch above

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

mrMoe, 2020-09-16:
I can confirm the fix in Arch Linux rtfm 5.8.8-arch1-1 #1 SMP PREEMPT Wed, 09 Sep 2020 18:59:45 +0000 x86_64 GNU/Linux

```
$ fwupdmgr --version                                                                                                                                                                                                        client version: 1.4.6
compile-time dependency versions
        gusb:   0.3.5
        efivar: 37
daemon version: 1.4.6
```

```
├─System Firmware:
│     Device ID:          fb5523b663445df99fe178d162a221546e1974af
│     Current version:    66816
│     Minimum Version:    66816
│     Vendor:             Dell Inc. (DMI:Dell Inc.)
│     GUID:               20448dae-1dd2-5653-a2b3-c81b77007d2f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
```

closed 2020-09-16 by superm1