fwupd/firmware-dell #26: Dell XPS 7390 can't update firmware

cardonator, 2020-06-04:
Hi all, I'm trying to update the firmware on my Dell XPS 7390. I am on Kubuntu 20.04. I can see a firmware update in Discover but whenever I try to update using that, it just disappears and does nothing.

So then I switched to trying to do it with `fwupdmgr`. It can find the firmware, download it, and prompt to reboot. However, on reboot the Dell logo just flashes and I get back to Kubuntu on the same old version of the BIOS I was on before.

I turned on debug logging and this was the output:
```
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b-0
../plugins/uefi/efi/fwupdate.c:577:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)/\EFI\ubuntu\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\ubuntu\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:415:fwup_add_update_capsule(): Read file; 226670 bytes
../plugins/uefi/efi/fwupdate.c:416:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../plugins/uefi/efi/fwupdate.c:417:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../plugins/uefi/efi/fwupdate.c:434:fwup_add_update_capsule(): Checking GOP for ux capsule
../plugins/uefi/efi/fwupdate.c:577:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)/\EFI\ubuntu\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and PciRoot(0)/Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Pci(0x1D,0x4)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668) and HD(Part1,Sig0FFC38D4-4E2D-42F1-A9FE-82F0DCC65668)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\ubuntu\fw\fwupd-cbe49cca-492b-93e8-b0f4-f693501f271b.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:415:fwup_add_update_capsule(): Read file; 12222506 bytes
../plugins/uefi/efi/fwupdate.c:416:fwup_add_update_capsule(): updates guid: CBE49CCA-492B-93E8-B0F4-F693501F271B
../plugins/uefi/efi/fwupdate.c:417:fwup_add_update_capsule(): File guid: CBE49CCA-492B-93E8-B0F4-F693501F271B
../plugins/uefi/efi/fwupdate.c:593:efi_main(): n_updates: 2
../plugins/uefi/efi/fwupdate.c:461:fwup_apply_capsules(): QueryCapsuleCapabilities: Success max: 104857600 reset:1
../plugins/uefi/efi/fwupdate.c:463:fwup_apply_capsules(): Capsules: 2
```

I didn't see anything helpful to what's going on there.

I then checked with `fwupdmgr get-history` and got this output.

```
XPS 13 7390
│
└─System Firmware:
  │   Device ID:           b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  │   Previous version:    1.1.3
  │   Update State:        needs-reboot
  │   Update Error:        failed to update to 0: invalid firmware format
  │   Last modified:       2020-06-04 16:21
  │   GUID:                cbe49cca-492b-93e8-b0f4-f693501f271b
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─XPS 13 7390 System Update:
        New version:       1.5.1
        Remote ID:         lvfs
        Summary:           Firmware for the Dell XPS 13 7390
        License:           Proprietary
        Size:              12.2 MB
        Vendor:            Dell Inc.
        Flags:             is-upgrade
        Description:       This stable release fixes the following issues:
        
         • Firmware updates to address security advisory
         • Firmware updates to address the Intel Security Advisories   and  .
         • Firmware updates to address the Intel security advisories
         • Firmware updates to address Intel security advisories  ,  , and  .
         • Fixed a BIOS Setup configuration issue that occurs after clearing the CMOS.
        
        Some new functionality has also been added:
        
         • This update integrates the BIOSConnect feature into Dell SupportAssist OS Recovery. This feature connects the system to the Dell image server to download and recover the operating system.
        
        Security issues fixed:
        
         • CVE-2019-14598
         • CVE-2019-11157
         • CVE-2019-14607
         • CVE-2019-11157
         • CVE-2019-0124
         • CVE-2019-0154
         • CVE-2019-0185
         • INTEL-SA-00307
         • INTEL-SA-00289
         • INTEL-SA-00317
         • INTEL-SA-00289
         • INTEL-SA-00220
         • INTEL-SA-00260
         • INTEL-SA-00254
```

I'm not really sure where to go from here. The firmware is obviously found but then fails to update. I don't know what `invalid firmware format` means or how to fix it and I can't find any results on Google that are helpful either.

Help? What other info can I provide? Thanks!

cardonator, 2020-06-04:
I haven't disabled secure boot so I am guessing the answer is yes? How do I determine this?

superm1, 2020-06-04:
Did this ship with Windows or Ubuntu?  If it shipped with Windows then yes it's turned on by default.  If you can please try to turn it off in BIOS setup to see if it helps that would be helpful.

If it shipped with Linux, it's most likely turned off, but you can confirm in BIOS setup.

cardonator, 2020-06-04:
Okay let me check that.

cardonator, 2020-06-04:
Switching off Secure Boot does seem to have worked. Thanks!

closed 2020-06-04 by cardonator

superm1, 2020-06-04:
actually this was reported previously and stalebot closed it, sorry!  It is not yet fixed.
https://github.com/fwupd/firmware-dell/issues/18