fwupd/firmware-dell #28: WD19/WD19DC/WD19TB firmware stuck in testing

sitnik, 2020-06-15:
Firmware update '01.00.14.01, A03' for WD19TB is generally available on Dell website, but in LVFS it stuck in 'testing' repository.

* Link to firmware update at OEM website
https://www.dell.com/support/home/en-us/product-support/product/dell-wd19-130w-dock/drivers
```
Version: 01.00.14.01, A03
Last Updated Date: 22 May 2020
```

* Firmware release date (if available on OEM website)
22 May 2020

* Link to the device page on LVFS.
https://fwupd.org/lvfs/devices/com.dell.wd198ceeeffd.firmware
```
Version 43.00, 01.00.14.01, 01.00.00.04, 01.47, 01.21, 05.04.03:
Warning: This firmware is in the testing state and may not be suitable for production systems.
Uploaded 2020-03-04 16:20:41.022858+00:00
```
It was uploaded earlier than 'Last Updated Date' on Dell site.

Please propagate this FW update to stable.

I can test upgrading it from testing and send a report if you want.

Also, a device name in LVFS is wrong, it doesn't have WD19TB in it, thus you can't find WD19TB by searching LVFS website. Should be changed from "Dell Inc. WD19/WD19DC/WDTB" to "Dell Inc. WD19/WD19DC/WD19TB".

superm1, 2020-06-15:
Yes it's currently intended to be in testing not stable.  Please if you have any problems with the update, do file a bug report though using this template: https://github.com/fwupd/fwupd/issues/new?assignees=superm1&labels=bug&template=bug-report-wd19.md&title=Dell+WD19+upgrade+issue

@hughsie is it possible to change the update title without redoing the upload?  I don't think so.  I would hate to lose all the metadata related to the update just to fix that.

hughsie, 2020-06-15:
> is it possible to change the update title without redoing the upload

You, no. Me, yes :) I've changed this for you on both public firmwares. It'll take ~2h to regen the metadata.

superm1, 2020-06-15:
>You, no. Me, yes :) I've changed this for you on both public firmwares. It'll take ~2h to regen the metadata.

Great thanks!  I'll get the thing that makes those XML files fixed for any future upload.

>It was uploaded earlier than 'Last Updated Date' on Dell site.

Well this actually is a tidbit that LVFS is leaking and probably shouldn't be.  That is the date that the file was first uploaded to LVFS, not the date that it was first made available to users.  A keen eye such as yours can see that these two dates are not the same.  I don't feel any conclusions can/should be drawn on this information.
@hughsie on this public facing page I think we should be showing a "First released publicly" date that shows the first movement to testing or stable not the date it was uploaded .

hughsie, 2020-06-15:
> I think we should be showing a "First released publicly" date that shows the first movement to testing or stable not the date it was uploaded

Agree. Give me a couple of mins to build a PR.

hughsie, 2020-06-15:
See https://github.com/fwupd/lvfs-website/pull/659

sitnik, 2020-06-15:
> Yes it's currently intended to be in testing not stable. Please if you have any problems with the update, do file a bug report though using this template: https://github.com/fwupd/fwupd/issues/new?assignees=superm1&labels=bug&template=bug-report-wd19.md&title=Dell+WD19+upgrade+issue

What is the process for transitioning to stable from testing?
Should some brave men install it from testing and report back?

sitnik, 2020-06-16:
Here is the ticket for upgrade issue: https://github.com/fwupd/fwupd/issues/2185
This one can be closed.

closed 2020-06-16 by sitnik