fwupd/firmware-dell #29: TB16 Thunderbolt Dock

maxamillion, 2020-06-21:
I see there is WD15 and WD19 docking station support, I was hoping I could request the addition of firmware updates for the TB16 dock also. 

Thank you!

hughsie, 2020-06-21:
@superm1 might know the plan here.

superm1, 2020-06-21:
One of the components in the TB16 is not upgradable from fwupd currently. So no it can't be supported at the moment.

hughsie, 2020-06-21:
Could we update any other bits of it?

superm1, 2020-06-21:
Yeah, it's technically possible to update the other bits, but I don't want to set the wrong expectations on the result from doing so...

hughsie, 2020-06-21:
Can you say the part without support? No is a valid option. 🙂

superm1, 2020-06-21:
https://github.com/fwupd/fwupd/issues/1351

hughsie, 2020-06-22:
Eww, the portio stuff. I have no idea why you'd use portio to update a USB controller. I still think it makes sense to update the other stuff -- e.g. nvm updates for compatibility, but I completely understand if Dell doesn't want to support a weird not-complete version update package from a QA point of view. Is there any way to set up a call with anyone at ASMedia to try and resolve this? I'm guessing they'd be super hesitant to do so with NDAs set up etc, and that's not going to happen from me.

superm1, 2020-06-22:
>I have no idea why you'd use portio to update a USB controller

It's specifically a USB host controller.

> but I completely understand if Dell doesn't want to support a weird not-complete version update package from a QA point of view

Yeah.

> Is there any way to set up a call with anyone at ASMedia to try and resolve this? I'm guessing they'd be super hesitant to do so with NDAs set up etc, and that's not going to happen from me.

This is getting more into the business side of things.  By chance have you seen this ASMedia part show up in anything else (other docks, other host systems etc)?

hughsie, 2020-06-22:
> a USB host controller.

In the dock itself? I'm no expert, but I have no idea how that is proxied through to an external device...

>  By chance have you seen this ASMedia part show up in anything else (other docks

Not on Lenovo or HP.

> , other host systems etc)?

From a firmware update point of view, unfortunately not. I guess in an ideal world we could convince them to do portio as an EFI binary (update capsule) but I'm not sure how that would work with the ESRT and it being a dock.

hughsie, 2020-06-22:
> I have no idea how that is proxied through to an external device

Doh, it's a TBT dock, so PCIe I guess.

superm1, 2020-06-22:
>Doh, it's a TBT dock, so PCIe I guess.

Yes..  

>Not on Lenovo or HP.

I would expect that other manufacturers in the Thunderbolt dock space use this as well, at least for their first gen Thunderbolt docks.  But the architecture is changed in the second generation Thunderbolt docks from Dell's side.

>From a firmware update point of view, unfortunately not. I guess in an ideal world we could convince them to do portio as an EFI binary (update capsule) but I'm not sure how that would work with the ESRT and it being a dock.

Yeah that would mean either needing to have BIOS support for that FMP already there, or it would mean putting it into something _like_ fwupdx64.efi.  I think the latter is _more_ achievable (like we envisioned for superio to transition to).

hughsie, 2020-06-22:
> mean putting it into something like fwupdx64.efi

Right. That isn't a "me" thing, that's right in the scope of of the consulting companies I think.

superm1, 2020-06-22:
If there was already an open source asmedia port io implementation to start off of that's one thing.
So even before operational coding work, like I said this is a business discussion type thing.

sorryusernameisalreadytaken, 2024-03-26:
This a great dock but useless without a chance to update the firmware.

I try it for 6 hours now to update the fw from 4 docks to get at least mac-os support but no chance with windows 10.

Is there any chance after almost 4 years after the ticket was created to see a possible to support for that Dock?

lostever, 2024-11-23:
> This a great dock but useless without a chance to update the firmware.
>
> I try it for 6 hours now to update the fw from 4 docks to get at least mac-os support but no chance with windows 10.
>
> Is there [...]

i have the same problem