fwupd/firmware-dell #31: Dell Inspiron 7559 System BIOS not in LVFS

barambani, 2020-07-01:
I see that there is no BIOS published for Dell Inspiron 15 7559 laptop on LVFS. There is a pretty important update [here](https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=c46fm&oscode=wt64a&productcode=inspiron-15-7559-laptop) so I was wondering if there's any chance to have it there, even if that machine doesn't have any official support for linux ? Thanks.

superm1, 2020-07-01:
@barambani for now you can follow the stuff in the README.md in this repo to update it.

barambani, 2020-07-01:
Thanks for the quick reply @superm1 , I tried but getting this
```
Traceback (most recent call last):
  File "./contrib/firmware_packager/install_dell_bios_exe.py", line 13, in <module>
    gi.require_version('Fwupd', '2.0')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Fwupd not available
```

superm1, 2020-07-01:
@barambani do you have fwupd installed on your system?  That means the GObject introspection libraries aren't available in your distro.

barambani, 2020-07-01:
Yes I do. I have 1.3.11. Also I can list the devices. I'm on Kali though, not Ubuntu. Not sure if it matters.

superm1, 2020-07-01:
It's possible Kali had a packaging problem and didn't ship the gobject introspection symbols.  They're definitely a requirement for this to work.

barambani, 2020-07-01:
Makes sense, thanks for the hint. I'll investigate. About this issue, do you think there are any chances that this firmware will make it into LVFS ? Let me know if there's no point in keeping this open. Thanks again.

superm1, 2020-07-01:
The tracker is intended for platforms that had firmware on LVFS that was missing an update, not for platforms without firmware on LVFS.

closed 2020-07-01 by superm1