fwupd/firmware-dell #32: TPM PCR0 differs from reconstruction after BIOS update

ppwaskie, 2020-08-10:
I'm running a Dell XPS 9300 laptop with Ubuntu 20.04 and Windows 10.  I just updated my BIOS to Dell's 1.1 image, and on the subsequent boot successfully updated it.  However, after booting Linux, I now see TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction in my Ubuntu Software update app.

See attached for the output from fwupdtpmevlog.  Not sure how to resolve this.  I'm also running 1.3.11 of fwupd.

[foo.txt](https://github.com/fwupd/fwupd/files/5014896/foo.txt)

ppwaskie, 2020-08-10:
> > in my Ubuntu Software update app
> 
> Can you show a screenshot where you're seeing it? It's surprising that it would be in the app. It should have been moved into the journal if it still exists.

See attached images.  Happy to provide anything else if I can.

[image]

[image]

superm1, 2020-08-10:
Oh... I see what's going on.  I got confused because @hughsie threw me off when mentioning 57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e is already in 1.3.11.  It indeed is, but the code change we made to make this not pop up in `UpdateError` landed in https://github.com/fwupd/fwupd/commit/1b73a132461ca15d4aadf4c0cfd2b5e95ac14d02, which isn't in a tagged version.

And you're seeing this from GNOME Firmware, not from Software Center, so that explains the other confusion. 

But yes, I feel this is likely a firmware issue at this point (unless we happen to have _another_ problem in the code, but I think that's unlikely now).

ppwaskie, 2020-08-10:
Thanks for the quick update and reply.

Should I open an issue then with Dell?

superm1, 2020-08-10:
If you're inclined to do so, yes putting this on tech support radar would be helpful in getting it investigated and fixed if necessary.

ppwaskie, 2020-08-11:
@superm1 is this likely the same or similar bug to this: https://github.com/fwupd/firmware-dell/issues/20 ?

superm1, 2020-08-11:
@ppwaskie #20 is missing fixes specifically in 1.3.11.  There was a problem in the tool as mentioned in https://github.com/fwupd/firmware-dell/issues/20#issuecomment-644401084

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

superm1, 2021-02-01:
We're fairly certain this is fixed in current fwupd.  If you can still reproduce with a current fwupd release, please open a new issue and link back to this one.

closed 2021-02-01 by superm1