fwupd/firmware-dell #33: XPS15: TPM PCR0 differs from reconstruction

fdr, 2020-08-13:
**Describe the bug**

running fwupdtool update yields me the following message. This is new behavior to me:

```
05:19:26:0008 FuPluginTpmEventlog  TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
```
I followed the link and it seems to be suggesting I report something.

**Steps to Reproduce**

run `sudo fwupdtool update`

**Expected behavior**

I don't expect a warning when running the update.

**fwupd version information**

```shell
$ fwupdmgr --version
```
```
fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```

This is a package included with Fedora 32.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

```
XPS 15 7590
│
├─Thunderbolt Controller:
│     Device ID:           8b19c110c0de2e30ec55fe70f0ef2e5e6cd5cc09
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     41.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               cabd17f2-67af-5a15-812c-e36de87d6133 ← THUNDERBOLT\VEN_00D4&DEV_0905&REV_00
│                          87ef6395-d440-5709-b471-1e152117c8a0 ← THUNDERBOLT\VEN_00D4&DEV_0905
│                          dc6ebd31-1fe5-5230-8090-dd0b06da4f87 ← TBT-00d40905-native
│                          dd244205-fa73-594e-b5ce-3aab372825e0 ← TBT-00d40905-native-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     12.0.68.1606
│     Bootloader Version:  12.0.68.1606
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
│     Device Flags:        • Internal device
│   
├─Intel(R) Core™ i7-9750H CPU @ 2.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xd6
│     Vendor:              GenuineIntel
[... 719 lines cut ...]
ChecksumSha256:          53018b830df5f594666497388d2884a2b68f27fe78256d3c2004738b56ce3121
BlobStr:                 (hd0,gpt7)/initramfs-5.7.14-200.fc32.x86_64.img.

PCRs:
  0:                     a42956fe598b7cbb07b84ae4a8156f39c8cfe800
  0:                     7b6af2045ec84be3dcb62f89b7fbb43f5511ca5f8ac78609f3c67b8e0383d1c8
  1:                     9cabd39e2cbbd3aa121f0668d5971bb1a419169b
  1:                     3d633ca1be7cc70e8624d87c258e1e98c71c4bf778e184cc9a9d9ce59fb1361e
  2:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  2:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
  3:                     b2a83b0ebf2f8374299a5b2bdfc31ea955ad7236
  3:                     3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
  4:                     d7694d85fdddc77d18cdd0267890f1709cbb845c
  4:                     b57f12630f377fb02872b041f2d6b7effa73400a1fd901271c25ed23b4f6a17d
  5:                     794d3ffd43738a19f29f47912ffa77ba5656ae62
  5:                     0a7eebbeefa0d176f049e27cd9b806b2dc426c7147006ced729e1a6f3b763a8c
  6:                     efd948b93289fb01a8f819766cf9b170bf074bc5
  6:                     a9753b412b1b6a9232a94c7814025ad19e9d52a80f9ae3270edf8d8246ae4ac6
  7:                     0543c8837092246b5659e7f7994da3eeb397304f
  7:                     1d08794a13b73f33b427345e6172e42fe879005ac1daee34834291025b9ea66d
  8:                     7181a74f6e1d10b040468a2fcbe9b9add518e4cc
  8:                     505b8045da36182de5777d60c53c6baada732cb214b936f3beb3c619d28e617b
  9:                     db077231c79a9837193ef4485baa0d6e61b99616
  9:                     b8723ab249cd576635c3df39e331e6687ff3403e6b733a324b85ed4825efc763
```

superm1, 2020-09-15:
A new tool bug was recently identified and fixed in master, 1_4_x, and 1_3_X branches.  It's not in any released version yet, but will be in 1.5.0 from master, 1.4.7 from 1_4_X and 1.3.12 from 1_3_X in the future.
https://github.com/fwupd/fwupd/pull/2394

Please upgrade to a new version with the patch integrated to confirm if this bug still exists.

superm1, 2021-02-01:
We're fairly certain this is fixed with current fwupd.  if you encounter this again, please reproduce with current fwupd and open a new issue linking back to this one.

closed 2021-02-01 by superm1

fdr, 2021-02-01:
Indeed: no more

```
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5
```