fwupd/firmware-dell #34: Dell PowerEdge Server: Construction of PCR0 using TPM event log is failing

prbinu, 2020-08-13:
**Describe the bug**
Background: https://github.com/tpm2-software/tpm2-tools/issues/1975#issuecomment-666719973

The construction of PCR0 using TPM event log in a tool called fwupdtpmevelog.

**Steps to Reproduce**

```
$ sudo fwupdmgr security --force
Host Security ID: HSI:0+! (v1.5.0)

HSI-1
✔ AMT manufacturing mode:        Locked
✔ AMT override:                  Locked
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✘ UEFI dbx:                      Not found: https://github.com/fwupd/fwupd/wiki/Missingdbx

HSI-2
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✘ IOMMU:                         Not found
✘ Intel DCI debugger:            Unlocked
✘ TPM PCR0 reconstruction:       Not found

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Suspend-to-ram:                Disabled
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

```bash
$ sudo  tsseventextend -sim -if /sys/kernel/security/tpm0/binary_bios_measurements 
eventextend: failed, rc 0000009a
TPM_RC_INSUFFICIENT - the TPM was unable to unmarshal a value because there were not enough octets in the input buffer Handle number unspecified
````

**Expected behavior**

The construction of PCR0 using TPM event log should match actual digest in  TPM PCR0

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --filter=~internal
PowerEdge R740
```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [ ] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
No

**Verbose daemon logs**
First enable daemon verbose logs collection.
```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.
```shell
journalctl -b -u fwupd.service
```

**Additional questions**
- Operating system and version: Ubuntu 20-04
- Have you tried unplugging the dock or any peripherals from your machine? No
- Have you tried to power cycle the dock from the AC adapter? No
- Is this a regression?

charles-rose, 2020-08-13:
I just got a system to recreate this issue. I will update this tracker what I find.

prbinu, 2020-08-14:
Observed this error, so sharing it:
" Update Error:        UEFI Capsule updates not available or enabled"

```
$ fwupdmgr get-devices
PowerEdge R740
│...
│
└─System Firmware:
      Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
      Current version:     2.5.4
      Vendor:              Dell Inc.
      Update Error:        UEFI Capsule updates not available or enabled
      GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
      Device Flags:        • Internal device
                           • Requires AC power
                           • Needs a reboot after installation```

charles-rose, 2020-08-18:
> Observed this error, so sharing it:
> " Update Error: UEFI Capsule updates not available or enabled"
>
> ```
> $ fwupdmgr get-devices
> PowerEdge R740
> │...
> │
> └─System Firmware:
>       Device ID: [...]

Dell Servers do not support Capsule updates. So this is expected.

charles-rose, 2020-08-18:
@prbinu Here is an interim update: I do not see the issue with tsseventextend when I set TPM to SHA256 in the BIOS setup. The default is SHA1 which seems to result in the TPM_RC_INSUFFICIENT error.

prbinu, 2020-08-19:
Thanks @charles-rose. Changing TPM from SHA1 to SHA256 resolve that issue.
I can now parse the event log. However when I tried to reconstruct and compare PCRs, I've observed that the computed value doesn't match with the PCR0 value. But rest of the PCRs are matching. Wondering what could go wrong here. Appreciate if you can guide us in right direction.

```bash
% sudo fwupd.fwupdtpmevlog -p 0
...
Reconstructed PCRs:
  PCR 0:                 SHA256(8027c3c86acfbb3f4710e45a66490ea660b1d587584d3cacefb72e188ba099f5)
  PCR 1:                 SHA256(10c6791fade2dc87c0208870837f5dd71611ebb257d2b878abcbb08b994e8f89)
  PCR 2:                 SHA256(e819672462843544d1d7c0f775b0d8e31cbd5f1c78bf1cb2d1b9ad570c53adce)
  PCR 3:                 SHA256(3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969)
  PCR 4:                 SHA256(c180eddf42eec74d21ae6f982413d9fb4bfa884eaf1f08fae2f76b6265550be8)
  PCR 5:                 SHA256(21938ffb3bc66d9c0a33d1592af8cf3a5b940e2eb138eeb45525e2237446e84a)
  PCR 6:                 SHA256(3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969)
  PCR 7:                 SHA256(f27be99e5d689a39134a2cbce032aa31ca5ce723531f50e1c4963e4b74c4e20f)
  PCR 8:                 SHA256(1abdd7000c35168d31bfc91c74630cc2313b2090bacdb1d81eb3c57af3416f31)
  PCR 9:                 SHA256(30136f90173eac52040cee846191681d621f84f83ca1f0caad35afae0432b02c)

% sudo TPM2TOOLS_TCTI=device:/dev/tpmrm0 tpm2_pcrread sha256:0,1,2,3,4,5,6,7,8,9,10
sha256:
  0 : 0x85BB2804189A1BF4C6228C90FCB8345102C938FA06092855AB22D56E1B04F2A0
  1 : 0x10C6791FADE2DC87C0208870837F5DD71611EBB257D2B878ABCBB08B994E8F89
  2 : 0xE819672462843544D1D7C0F775B0D8E31CBD5F1C78BF1CB2D1B9AD570C53ADCE
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0xC180EDDF42EEC74D21AE6F982413D9FB4BFA884EAF1F08FAE2F76B6265550BE8
  5 : 0x21938FFB3BC66D9C0A33D1592AF8CF3A5B940E2EB138EEB45525E2237446E84A
  6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  7 : 0xF27BE99E5D689A39134A2CBCE032AA31CA5CE723531F50E1C4963E4B74C4E20F
  8 : 0x1ABDD7000C35168D31BFC91C74630CC2313B2090BACDB1D81EB3C57AF3416F31
  9 : 0x30136F90173EAC52040CEE846191681D621F84F83CA1F0CAAD35AFAE0432B02C
  10: 0x0B73040EFF3E51712C605856191FB7BD26B985D9103890202A8B9CEAFE26CE96
```

freedge, 2022-07-27:
Same here. It first failed because sha1 is the default, changing to sha256 in the BIOS.
now tpm2_eventlog is working, but there the is an issue with pcr0. I notice this warning:

```
- EventNum: 3
  PCRIndex: 0
  EventType: EV_S_CRTM_CONTENTS
  DigestCount: 1
  Digests:
  - AlgorithmId: sha256
    Digest: "3b607ac1051c9f694a8983a68c9578ee1e5fedeeb8b1b9d8f8780068b7d10016"
  EventSize: 27
  Event:
    BlobBase: 0x61754720746f6f42
    BlobLength: 0x757361654d206472
- EventNum: 4
  PCRIndex: 0
  EventType: EV_S_CRTM_VERSION
  DigestCount: 1
  Digests:
  - AlgorithmId: sha256
    Digest: "ea7e50af2e8075aedecda37e783a583b0b4895dddf98140e9d92a052ace32a2d"
WARN: Event 4's digest does not match its payload
  EventSize: 14
  Event: "32002e00310034002e0032000000"

```

tomoveu, 2022-07-29:
@prbinu , @freedge could it be that PCR0 is mismatching because you changed your UEFI settings (sha1 to sha256) ...

... and now the CRTM measurement is failing because it is hashing the UEFI settings that have been modified from the default?

freedge, 2022-07-29:
(I have opened a case to Dell about this, but I'm not too sure it will reach the right team as it is very specific and probably far from what the L1 support gets usually)
I note that 32002e00310034002e0032000000 is "2.14.2" in ASCII which matches the BIOS version, and I can understand that people could want a stable PCR0 even if the BIOS is upgraded.

If you have any idea of a test I can run I can do that (but tpm2_eventlog is broken when sha1is selected)

Sinihopea, 2025-02-17:
> (I have opened a case to Dell about this, but I'm not too sure it will reach the right team as it is very specific and probably far from what the L1 support gets usually) I note that [...]

Hello,

has there been any progress with the Dell support on this issue in your case? I mean... regarding the "WARN: Event 3's digest does not match its payload"?

Thanks!