fwupd/firmware-dell #35: Unable to update "Dell Inc. System Firmware" as download failed

gkzz, 2020-09-08:
This message is the follwoing:

```
Unable to update "Dell Inc. System Firmware" as download failed:
failed to download https://fwupd.org/downloads/-xxxxxxx-d35-XPS_9380_1.11.0.cab: Connection terminated unexpectedlly
```
[image]

All I have to do is download the following file, isn't it??
https://fwupd.org/lvfs/devices/com.dell.uefice945437.firmware

hughsie, 2020-09-08:
Does it download correctly using `wget` or `curl`?

gkzz, 2020-09-08:
> Does it download correctly using `wget` or `curl`?

Thanks you for your quick comment!

- I've executed curl command.

```
 curl https://fwupd.org/downloads/60344cbcc1f3c061ea202e1c7d7cc9c6333f0f070eba09fc8ef27114fe3ced35-XPS_9380_1.11.0.cab --output ~/Downloads/xps_9380.cab
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 21.6M  100 21.6M    0     0  1454k      0  0:00:15  0:00:15 --:--:-- 2711k
```
- I'm successful in it
```
$ ls ~/Downloads/xps_9380.cab 
/home/gkz/Downloads/xps_9380.cab
```
Could you please share your opinion what is the next to do?
Should I run this cab file as follows:

```
$ . /home/gkz/Downloads/xps_9380.cab
```

superm1, 2020-09-08:
Can you try again in the GUI?  It seems like a transient network issue to me.

gkzz, 2020-09-08:
> Can you try again in the GUI? It seems like a transient network issue to me.

I appreciate it!

I got the following error message after I clicked `$HOME/Downloads/xps_9380.cab`.

```
Failed to install file: not supported
```

superm1, 2020-09-09:
I mean try to run the update again pulling from the network not opening the file.  Although that opening the file looks like a different bug..

gkzz, 2020-09-09:
Thanks for your quick response again.

I tried again over a wired connection, however I got the same problem. 
https://github.com/fwupd/firmware-dell/issues/35#issue-695888292

I have a question.
My question is my issue is the same as [#967](https://github.com/fwupd/fwupd/issues/967) , isn't it??
What do you think so, too?

The following may be good..
https://github.com/fwupd/fwupd/issues/967#issuecomment-515910658
https://github.com/fwupd/fwupd/issues/967#issuecomment-515929474
https://github.com/fwupd/fwupd/issues/967#issuecomment-515929770

superm1, 2020-09-09:
>I tried again over a wired connection, however I got the same problem.

That's bizarre to me that it worked properly in browser and is failing in Gnome Software.  Do you have a network proxy of some sort that might be influencing?

> My question is my issue is the same as #967 , isn't it??

I don't think so.  Looks like a different error, and that comes from fwupdmgr not gnome-software.

I think you can just run `fwupdmgr install $FILE` for now.

gkzz, 2020-09-10:
>That's bizarre to me that it worked properly in browser and is failing in Gnome Software. Do you have a network proxy of some sort that might be influencing?

Hmm. I don't use a proxy or anything. .. The network provider may be doing something.

gkzz, 2020-09-10:
All I have to do is as bellows:

- Get the following cab file
```
$ curl https://fwupd.org/downloads/60344cbcc1f3c061ea202e1c7d7cc9c6333f0f070eba09fc8ef27114fe3ced35-XPS_9380_1.11.0.cab \
> --output $HOME/Downloads/xps_9380.cab
```

- Execute `fwupdmgr install $FILE`

```
$ fwupdmgr install Downloads/xps_9380.cab 
```

- Type y, after the following is displayed.
```
$ fwupdmgr install Downloads/xps_9380.cab 
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on System Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: 
```

closed 2020-09-10 by gkzz