fwupd/firmware-dell #37: [Missing Firmware] Alienware m15

jhaygood86, 2020-09-28:
Available Firmware at Dell Support: https://dl.dell.com/FOLDER06291943M/1/Alienware_m15_m17_2.5.0.exe
Available Firmware on LVFS: None
System: Alienware m15 laptop

superm1, 2020-09-28:
Sorry to say, but Alienware isn't currently in the list of supported systems for LVFS.  You can look at the Dell under the LVFS vendors link:
https://fwupd.org/lvfs/vendors/

If you want to make a case to Dell supporting it, you should contact support to request it.

For now, if you want to upgrade using Linux you can follow the directions in [README.md](https://github.com/fwupd/firmware-dell) to try to upgrade manually from an EXE on Linux.  (If it works feel free to leave a note here for anyone else who comes along)

closed 2020-09-28 by superm1

jhaygood86, 2020-09-28:
I was asked to open this by Richard Hughsie here: https://blogs.gnome.org/hughsie/2020/09/28/20-million-downloads-from-the-lvfs/#comment-24747

superm1, 2020-09-28:
Right, understand. This repo is for tracking missing firmware for supported systems. Feel free to reference it when you would reach out to support, but adding new systems isn't in the scope for the repo.