fwupd/firmware-dell #38: Latest firmware for Dell XPS 15 9550

daljit97, 2020-11-19:
Hi, I was trying to update the firmware for my Dell XPS 15 9550 which is 1.14.0 (released in October). Checking on the website it says that the latest firmware should be available https://fwupd.org/lvfs/devices/com.dell.uefi124c207d.firmware but when I run fwupdmgr and tried to update it says that there are no updates available (my current bios version is 1.13.1 release in January). 
Is there anyway to update to the latest firmware?

**system info**

> Operating System: Fedora 32
> KDE Plasma Version: 5.18.5
> KDE Frameworks Version: 5.73.0
> Qt Version: 5.14.2
> Kernel Version: 5.8.16-200.fc32.x86_64
> OS Type: 64-bit
> Processors: 8 × Intel® Core™ i7-6700HQ CPU @ 2.60GHz
> Memory: 15.5 GiB of RAM
> 

**fwupd version information**

> client version: 1.4.5
> compile-time dependency versions
>         gusb:   0.3.4
>         efivar: 37
> daemon version: 1.4.5

superm1, 2020-11-19:
Can you please share your `fwupdmgr get-devices` output?

daljit97, 2020-11-19:
`fwupdmgr get-devices` outputs:

> XPS 15 9550
> │
> ├─System Firmware:
> │     Device ID:           2da31270d317b076424992de14a0f08ae373c137
> │     Current version:     1.13.1
> │     Minimum Version:     1.13.1
> │     Vendor:              Dell Inc. (DMI:Dell Inc.)
> │     GUID:                124c207d-5db8-4d95-bd31-34fd971b34f9
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Needs a reboot after installation
> │                          • Cryptographic hash verification is available
> │                          • Device is usable for the duration of the update
> │   
> ├─THNSN5512GPU7 NVMe TOSHIBA 512GB:
> │     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
> │     Summary:             NVM Express Solid State Drive
> │     Current version:     57DA4103
> │     Vendor:              Toshiba Corporation (NVME:0x1179)
> │     Serial Number:       36TS10BJT5ZV
> │     GUIDs:               e22c4520-43dc-5bb3-8245-5787fead9b63 ← NVME\VEN_1179&DEV_010F&REV_01
> │                          83991323-9951-5adf-b743-d93e882a41e1 ← NVME\VEN_1179&DEV_010F
> │                          7e2d1b9e-3c62-5ae2-b2ef-eb6f98c5ff4b ← THNSN5512GPU7 NVMe TOSHIBA 512GB
> │     Device Flags:        • Internal device
> │                          • Updatable
> │                          • Requires AC power
> │                          • Needs a reboot after installation
> │                          • Device is usable for the duration of the update
> │   
> ├─TPM 1.2:
> │ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
> │ │   Summary:             Platform TPM device
> │ │   Current version:     5.81.0.0
> │ │   Vendor:              Dell Inc. (PCI:0x1028)
> │ │   Update Error:        Updating disabled due to TPM ownership
> │ │   GUIDs:               9055a777-a92f-55da-8d34-6614d8f9df92 ← 06e4-1.2
> │ │                        ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
> │ │   Device Flags:        • Internal device
> │ │                        • Requires AC power
> │ │ 
> │ └─TPM 2.0:
> │       Device ID:         04d075f2971403e31c1dbdf3b38f709d7c2e749d
> │       Summary:           Alternate mode for platform TPM device
> │       Update Error:      mode switch disabled due to TPM ownership
> │       GUID:              75305a8e-596f-5a3b-9aea-ebee36dc585e ← 06e4-2.0
> │       Device Flags:      • Internal device
> │                          • Requires AC power
> │                          • Device is locked
> │     
> └─Touchpad:
>       Device ID:           f6c3ee2cab703108f932db259fd58e7f1134d0de
>       Current version:     1.6.1961136
>       Bootloader Version:  54.0.0
>       Vendor:              Synaptics (HIDRAW:0x06CB)
>       GUIDs:               1d6705d0-3fa6-58d2-bd0e-806ed6792c18 ← HIDRAW\VEN_06CB&DEV_7A13&REV_00
>                            2aa17264-7ef0-5cc5-9368-8595974a5e31 ← HIDRAW\VEN_06CB&DEV_7A13
>                            741692a4-ae19-5ecc-9e48-e644604c2345 ← SYNAPTICS_RMI\TM3125-001
>                            a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8 ← SYNAPTICS_RMI\TM3125
>       Device Flags:        • Internal device
>                            • Updatable

superm1, 2020-11-19:
Hmm.  It looks like the underlying component is correct (https://fwupd.org/lvfs/components/6572/overview).  The number of downloads is way lower than I would expect, so something does seem fishy.

Have you recently run `fwupdmgr refresh`?  At least in the last 30 days (when that was promoted to stable)?

After running refresh can you please provide output of this:

`fwupdtool update --verbose`

That should help capture the reason why it's not getting offered to you.

daljit97, 2020-11-19:
Here is the output of `sudo fwupdtool update --verbose`:

> 16:57:33:0323 FuDebug              Verbose debugging enabled (on console 1)
> 16:57:33:0327 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
> 16:57:33:0327 FuConfig             loading config values from /etc/fwupd/daemon.conf
> 16:57:33:0329 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
> 16:57:33:0329 FuRemoteList         loading /usr/share/fwupd/metainfo
> 16:57:33:0329 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
> 16:57:33:0329 XbSilo               file: 2c0213f0-ec89-774a-1958-dfcdf184436c, current:2c0213f0-ec89-774a-1958-dfcdf184436c, cached: (null)
> 16:57:33:0329 XbSilo               loading silo with file contents
> 16:57:33:0329 Fwupd                looking for os-release at /etc/os-release
> 16:57:33:0329 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
> 16:57:33:0329 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
> 16:57:33:0329 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
> 16:57:33:0329 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
> 16:57:33:0330 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
> 16:57:33:0330 FuRemoteList         ignoring unfound remote fwupd
> 16:57:33:0330 FuRemoteList         ordering lvfs-testing=lvfs+1
> 16:57:33:0330 FuRemoteList         ignoring unfound remote fwupd
> 16:57:33:0330 FuRemoteList         ignoring unfound remote fwupd
> 16:57:33:0330 FuRemoteList         ignoring unfound remote fwupd
> 16:57:33:0330 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
> 16:57:33:0330 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
> 16:57:33:0330 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
> 16:57:33:0330 Jcat                 loaded 1 certificates
> 16:57:33:0330 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
> 16:57:33:0330 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
> 16:57:33:0330 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
> 16:57:33:0330 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
> 16:57:33:0330 Jcat                 loaded 1 certificates
> 16:57:33:0330 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
> 16:57:33:0330 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
> 16:57:33:0330 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
> 16:57:33:0330 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
> 16:57:33:0337 FuEngine             client certificate exists and working
> 16:57:33:0337 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
> 16:57:33:0337 FuHistory            got schema version of 5
> 16:57:33:0337 FuHwids              smbios property Manufacturer=Dell Inc.
> 16:57:33:0337 FuHwids              smbios property EnclosureKind=9
> 16:57:33:0337 FuHwids              smbios property Family=XPS
> 16:57:33:0337 FuHwids              smbios property ProductName=XPS 15 9550
> 16:57:33:0337 FuHwids              smbios property ProductSku=06E4
> 16:57:33:0337 FuHwids              smbios property BiosVendor=Dell Inc.
> 16:57:33:0337 FuHwids              smbios property BiosVersion=1.13.1
> 16:57:33:0337 FuHwids              smbios property BiosMajorRelease=01
> 16:57:33:0337 FuHwids              smbios property BiosMinorRelease=0d                                                                                                                
> 16:57:33:0337 FuHwids              smbios property BaseboardManufacturer=Dell Inc.                                                                                                    
> 16:57:33:0337 FuHwids              smbios property BaseboardProduct=0Y9N5X                                                                                                            
> 16:57:33:0340 FuQuirks             no /var/lib/fwupd/quirks.d, skipping                                                                                                               
> 16:57:33:0340 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb                                                                                                    
> 16:57:33:0340 XbSilo               file: 50ad20dc-5924-7e6f-d38e-cab2580af487, current:50ad20dc-5924-7e6f-d38e-cab2580af487, cached: (null)                                           
> 16:57:33:0340 XbSilo               loading silo with file contents                                                                                                                    
> 16:57:33:0340 FuEngine             remote lvfs-testing not enabled, so skipping                                                                                                       
> 16:57:33:0340 FuEngine             remote vendor not enabled, so skipping                                                                                                             
[... cut ...]
> 16:57:34:0270 FuPlugin             performing destroy() on flashrom
> 16:57:34:0270 FuPlugin             performing destroy() on modem_manager
> 16:57:34:0270 FuPlugin             performing destroy() on synaptics_mst
> 16:57:34:0270 FuPlugin             performing destroy() on upower
> 16:57:34:0271 FuPlugin             performing destroy() on dell
> 16:57:34:0271 FuPlugin             performing destroy() on redfish
> 16:57:34:0271 FuPlugin             performing destroy() on uefi
> 16:57:34:0271 FuPlugin             performing destroy() on logind

daljit97, 2020-12-04:
Any update on this?

superm1, 2020-12-10:
Can you please check if you have udisks2 installed?  If you don't have it installed, do you have `/etc/fwupd/uefi.conf` with a value set for ESP path?

daljit97, 2020-12-10:
@superm1 yes, `udisks2-2.8.4-4.fc32.x86_64` is installed.

superm1, 2020-12-10:
Okay in the linked issue I think we figured out what's going on. Can you please download the cab manually and install it?

daljit97, 2020-12-11:
@superm1 that worked! In future, will I have to do this manually every time there is an update?

hughsie, 2020-12-11:
> will I have to do this manually every time there is an update

No, we're debugging what went wrong now.