fwupd/firmware-dell #50: Dell XPS 13 9310 2-in-1: Missing firmware on LVFS

paulmenzel, 2021-02-01:
I am unable to find firmware updates for the Dell XPS 13 9310 2-in-1 on LVFS: https://fwupd.org/lvfs/search?value=9310.

     $ sudo dmesg | grep DMI
     [    0.000000] DMI: Dell Inc. XPS 13 9310 2-in-1/02P55J, BIOS 1.0.3 10/05/2020

[The current version is 2.0.0.](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=228t8&oscode=wt64a&productcode=xps-13-9310-2-in-1-laptop)

Maybe the reason is, that it’s not officially shipped with GNU/Linux?

```
$ fwupdmgr --version
client version:	1.5.5
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.5
$ fwupdmgr get-devices
XPS 13 9310 2-in-1
│
├─KBG40ZPZ256G NVMe KIOXIA 256GB:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express Solid State Drive
│     Current version:    10400105
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              fbef9958-2179-5fb1-9e91-31c22cc29b2d
│                         b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─System Firmware:
│ │   Device ID:          3799e725ec98d58dcf68eea254c50ef7dd261a4f
│ │   Current version:    65539
│ │   Minimum Version:    65539
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              a472a89c-42ab-c171-228e-e182e424af76
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       055d2140-2ac7-5568-ae08-d78b141978ae
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─Touchpad:
│     Device ID:          6042b531410f9a5c52c89376621f5382911db5ab
│     Current version:    1.3.3260652
│     Bootloader Version: 54.0.0
│     Vendor:             Synaptics (HIDRAW:0x06CB)
│     GUIDs:              c9d02f56-521a-5797-8842-255abc8b9187
│                         613c9012-bcb6-5809-9aaa-8e244cf2de0e
│                         bbfe9e12-4f9d-5bb7-b1fa-56c05ace5393
│                         181b0f39-26f3-54a8-a2b4-ceb67afc791b
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          6b9d757c1f64d696e7af69942c2d6ed3993479d4
│     Current version:    96
│     Minimum Version:    96
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              176e090d-0ddb-495e-8173-bc998ccfecd0
│                         3e1d83c5-31af-55ac-9f88-8be02bda9596
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          7e3dfff3a5ac665b2b413a1a1a3babdadc1c3784
│     Current version:    272630021
│     Minimum Version:    272630021
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              b2cca4ac-d65f-e811-9c2d-fa7ae01bbebc
│                         0360499c-14d8-53b8-938b-37de40587af4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Wacom AES Device:
      Device ID:          dbbf40284cb602275c039924b932efc748233670
      Current version:    3006.00
      Vendor:             HIDRAW:0x056A
      GUIDs:              f2e31e0f-ab3b-5cd2-8a2f-462ada7ed50a
                          5f02cd21-7586-5650-95e9-388fdf0b6e63
                          a6531f36-4cd5-5878-a8f4-9efc1b7702b3
      Device Flags:       • Internal device
                          • Updatable

$ fwupdmgr update
Devices with no available firmware updates: 
 • KBG40ZPZ256G NVMe KIOXIA 256GB
 • System Firmware
 • Touchpad
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • Wacom AES Device
No updatable devices
```

superm1, 2021-02-01:
This repository is specifically for machines that already have firmware on LVFS.  Since this system doesn't, the best suggestion is as in the README to install with the helper script (which I'm aware you tried and have a problem, we'll figure that out in https://github.com/fwupd/fwupd/issues/2836)

closed 2021-02-01 by superm1

paulmenzel, 2021-02-01:
As always thank you for your support. Where should I report the issue to start getting the firmware files added to LVFS?

superm1, 2021-02-01:
You can request in the Dell forums or with Dell support and they would need to prioritize the request with the appropriate internal teams.

bionade24, 2021-02-03:
The thing in the README.md doesn't work for me :(. `Couldn't find any UEFI devices` @paulmenzel Have you already reached out to Dell somewhere?

paulmenzel, 2021-02-03:
> The thing in the README.md doesn't work for me :(. `Couldn't find any UEFI devices`

Unfortunately, the laptop is in production use by another user. Please report a separate issue for that at https://github.com/fwupd/fwupd/issues/.

> @paulmenzel Have you already reached out to Dell somewhere?

Yes, I did. Here is the negative reply (the second has at typo and I guess means *nicht Hochladen*:

### my reply to first reply ###

Am 02.02.21 um 16:20 schrieb Dell Tech Support:
> Am 02.02.21 um 15:29 schrieb Dell Tech Support:
> 
>> Wenn ich Sie richtig verstehe wollen Sie das BIOS unter Linux updaten ?
> im Prinzip ja, obwohl das technisch nicht ganz korrekt ist. fwupd
> unterstützt Aktualisierungen im UEFI-Capsule-Format. Dieselbe Datei, die
> auch „unter Windows“ benutzt wird, wird in den Arbeitsspeicher geladen
> und beim Neustart von der Firmware erkannt.
> 
>> Linux ist für dieses System nicht ausgelegt und nicht von Dell
>> getestet wurden.
> Korrekt. Leider bietet Dell das System nur mit Microsoft Windows an.
> Trotzdem konnten wir erfolgreich eine GNU/Linux-Distribution darauf
> installieren.
> 
>> Sie können ein BIOS update über das Boot Menü und USB stick
>> durchführen.
> Ich kann das, es ist aber umständlicher und nicht nachvollziehbar. Zudem
> kann auch das Python-Skript mit der heruntergeladenen Exe-Datei
> ausgeführt werden [1]. Das Problem ist, dass das alles nicht automatisch
> funktioniert und die Nutzer*innen nicht über Aktualisierungen informiert
> würden.
> 
>> Sollte ich hier falsch liegen bitte geben Sie mir bescheid und wir
>> können das Thema gerne besprechen.
> Dell müsste die Firmwaredatei zu LVFS hochladen, wie von Ihrem
> Entwickler bei GitHub geschrieben [2]:
> 
>> You can request in the Dell forums or with Dell support and they
>> would need to prioritize the request with the appropriate internal
>> teams.
> 1. Bitte beachten Sie auch, dass dasselbe Problem das Gerät mit
> Service-Tag GMJBGB3 betrifft. Soll ich dafür eine separate Anfrage
> erstellen, damit der Bedarf in Dells Ticketsystem auftaucht (sich
> verdoppelt ;-))?
> 
> 2. Wenn ich zu der Firmware-Seite *System-BIOS für Dell XPS 13
> 9310-2-in-1-Systeme* [3] gehe, steht dort unter *Unterstützte
> Betriebssysteme*:
> 
>> BIOS
>> Looking for a different OS? View the list of Dell supported operating systems
> Der Verweis führt zu [4], wo steht:
> 
>> Informationen, Tutorials und Anleitungen zum Troubleshooting für die
>> von Dell unterstützten Betriebssysteme
>> Linux-Betriebssystem
> Die Bildschirmfotos sind angehängt.
> 
> 
> Freundliche Grüße

> 
> [1]: https://github.com/fwupd/firmware-dell/#updating-from-an-exe-without-a-capsule
> [2]: https://github.com/fwupd/firmware-dell/issues/50#issuecomment-770989977
> [3]: https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=228t8&oscode=wt64a&productcode=xps-13-9310-2-in-1-laptop
> [4]: https://www.dell.com/support/home/de-de/drivers/supportedos/xps-13-9310-2-in-1-laptop

### second reply ###

> Vielen Dank für die Rückmeldung.
> 
> Da dieses System nicht mit Linux ausgeliefert wird und nicht unterstützt wird, kann Dell die LVFS Datei Hochladen.
> Das andere System ist das Baugleiche System und wird auch nicht Linux unterstützt.
> 
> Ja es steht bei dem BIOS update auch Linux als unterstützte Betriebssystem, dies ist aber ein Downloadvorlage die von den > Herstellern der Webseite verwendet wird.

So, please also open a support request. Maybe their ticket system analysis is going to show the demand.

superm1, 2021-02-03:
> The thing in the README.md doesn't work for me :(. `Couldn't find any UEFI devices` @paulmenzel Have you already reached out to Dell somewhere?

Make sure you are installed in UEFI mode and have UEFI capsule update turned on in BIOS setup.

paulmenzel, 2021-07-23:
Noticing issue https://github.com/fwupd/firmware-dell/issues/72 (XPS 13 9310 2-in-1 is missing latest update addressing critical security vulnerabilities), Dell did the right move and seems to have started providing firmware updates for that device.