fwupd/firmware-dell #51: WD SN730 SSD Firmware update for Dell XPS 9310

sitnik, 2021-02-01:
Dell released a FW update for WD SN730 SSDs found in XPS 9310 and probably other laptop models:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=50v6h&oscode=wt64a&productcode=xps-13-9310-laptop

fwupdmgr sees this device and considers it updatable:
```
$ sudo fwupdmgr get-devices
XPS 13 9310
│
├─Integrated Webcam HD:
│     Device ID:           65a54fb6ce182f0e75edf0e43047d547a0d61f0e
│     Current version:     4.21
│     Vendor:              Integrated_Webcam_HD (USB:0x0BDA)
│     Serial Number:       200901010001
│     GUIDs:               df5723b8-9a6a-5071-9369-c7a64c7a2c8a ← USB\VID_0BDA&PID_5517&REV_0421
│                          2632c680-275f-54f7-8795-ecfaf34e14cf ← USB\VID_0BDA&PID_5517
│     Device Flags:        • Updatable
│   
├─PC SN730 NVMe WDC 512GB:
│     Device ID:           c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:             NVM Express Solid State Drive
│     Current version:     11121012
│     Vendor:              Sandisk Corp (NVME:0x15B7)
│     Serial Number:       203627803697
│     GUIDs:               10215a6e-5eb4-5fe4-a3d0-1ff4b3295840 ← STORAGE-DELL-108613
│                          e3a1928f-2cb0-4b92-8526-9368d8c0c2d1
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
...
```
fwupd.org even has a page for this device, but for another vendor laptops: https://fwupd.org/lvfs/devices/com.lenovo.PCSN730.firmware

Is there any chance to have SN730 updates on Dell systems via fwupdmgr?

superm1, 2021-02-01:
I'll reach around internally about this one to find out if there is a plan to release for Linux or not.

superm1, 2021-02-05:
@sitnik the SN730 should be up there now.

closed 2021-02-05 by superm1

sitnik, 2021-02-06:
I've updated this firmware and it went flawlessly. 
Thank you very much.