fwupd/firmware-dell #53: Dell XPS 13 7390: After reboot firmware update is *not* executed but drops in system check

paulmenzel, 2021-02-19:
On the Dell XPS 13 7390 with firmware 1.5.1, using Debian sid/unstable with fwupd 1.5.6, running `fwupdmgr update`, it downloads the update for UEFI firmware 1.7.0 and asks to reboot the system. It reboots, the screen is lit but nothing to see, then the speaker makes a cracking sound and it starts the UEFI firmware system diagnostics.

```
$ fwupdmgr --version
client version:	1.5.6
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.6
$ fwupdmgr get-history 
XPS 13 7390
│
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│ │   Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:   10604106
│ │   Update State:       Success
│ │   Last modified:      2021-02-19 12:44
│ │   GUID:               8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─XG6 NVMe Device Update:
│       New version:      10604107
│       Remote ID:        lvfs
│       Summary:          XG6 NVMe SSD firmware for Dell PC
│       License:          Proprietary
│       Size:             1.8 MB
│       Created:          2020-08-21
│       Urgency:          Medium
│       Vendor:           KIOXIA Corporation
│       Description:      
│       Optimize SMART thermal temperature implementation.
│     
└─System Firmware:
  │   Device ID:          b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  │   Previous version:   1.5.1
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2021-02-19 12:53
  │   GUID:               cbe49cca-492b-93e8-b0f4-f693501f271b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 13 7390 System Update:
        New version:      1.7.0
        Remote ID:        lvfs
        Summary:          Firmware for the Dell XPS 13 7390
        License:          Proprietary
        Size:             12.3 MB
        Created:          2020-10-22
        Urgency:          Critical
        Vendor:           Dell Inc.
        Flags:            is-upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Firmware updates to address the Intel Technical Advisory.
        • Fixed the issue where the Data wipe option does not show in the BIOS setup after switching between RAID and AHCI mode.
        • Fixed the issue where the external monitor does not power on it is restarted.
        • Fixed the issue where the Linux Vendor Firmware Service (LVFS) does not work when Secure Boot is enabled in the BIOS setup menu.
        
        Some new functionality has also been added:
        
        • Updated the Embedded Controller Engine firmware.
        • Updated the system Power Delivery firmware.
        • Updated Enhanced Pre-Boot System Assessment (ePSA) firmware.
        • Reduced the system fan noise.
        
        Security issues fixed:
        
        • INTEL-TA-00391
        • INTEL-SA-00389
        • INTEL-SA-00381
        • INTEL-SA-00358
        • CVE-2020-26189
        • CVE-2020-8695
        • CVE-2020-8694
        • CVE-2020-8698
        • CVE-2020-8696
        • CVE-2020-0593
        • CVE-2020-0592
        • CVE-2020-0591
        • CVE-2020-0587
        • CVE-2020-12355
        • CVE-2020-12303
        • CVE-2020-8755
        • CVE-2020-8705
        • CVE-2020-8744
        • CVE-2020-8745
```

paulmenzel, 2021-02-19:
No, Secure Boot is disabled in the firmware.

paulmenzel, 2021-02-19:
I am now updating the firmware from within the UEFI firmware (F12) and selecting the CAB file from the ESP (`EFI/debian/fw`).

superm1, 2021-02-19:
> No, Secure Boot is disabled in the firmware.

OK.  You didn't follow the bug template so there was no way to know.

>I am now updating the firmware from within the UEFI firmware (F12) and selecting the CAB file from the ESP (EFI/debian/fw).

Then there goes any stuff we can use for debugging.  I guess next firmware update if this persists please file a bug using the fwupd UEFI bug template.

paulmenzel, 2021-02-19:
Yes, sorry about that. Will do.

Here the more detailed observations:

1.  Reboot
2.  Dell logo is shown.
3.  Speakers make a cracking sound.
4.  UEFI firmware system diagnostics is started.

paulmenzel, 2021-02-19:
This device has an NVMe device, and Secure Boot is disabled.

And some output:

```
$ fwupdmgr get-devices --show-all-devices
XPS 13 7390
│
├─Thunderbolt host controller:
│     Device ID:          0434b3dc3d37a0178ef79214929996439d8402db
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    41.02
│     Vendor:             Dell (THUNDERBOLT:0x00D4|TBT:0x00D4)
│     GUIDs:              83f70f65-8de5-5520-93e2-a569615030bb
│                         b067bd0b-045e-5b00-9948-f7c98fef1f72
│                         7d538854-204d-51b2-8f9d-1fe881c70200
│                         8a1cafa7-2d8b-5641-ba4a-140a6497a5e9
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─USB3.1 Hub:
│     Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
│     Summary:            USB 3.x Hub
│     Current version:    9.153
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              30f80e42-579c-54b6-999a-db3d7378a63b
│                         08887a10-1663-55e7-ab55-a34cc4f0d011
│                         e2db8274-3ab0-5120-9d9a-2f9693cd7db8
│                         090da1fc-aefe-5e63-8b90-6e9f24780c7d
│                         9bed0c73-4ed7-5734-9919-1a217b2a63c0
│                         7e650fa0-f815-531c-be10-ad34374b4748
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─Intel(R) Core™ i7-10510U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000de
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         d5092916-1868-5182-8c70-12f83dab6372
│     Device Flags:       • Internal device
│   
├─KXG60ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│     Summary:            NVM Express Solid State Drive
│     Current version:    10604107
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     GUIDs:              8a412b95-5c07-5ce4-b63a-f9cbf254dd5d
│                         344c2804-4948-e811-842f-0ed5f89f718b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
│ │   Current version:    1.7.0
│ │   Minimum Version:    1.7.0
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              cbe49cca-492b-93e8-b0f4-f693501f271b
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
[... 41 lines cut ...]
│                         9779d679-0f31-5dd5-b3c2-f6ff275a9ca9
│                         5589d313-88e5-54d3-8f4c-b2ee3882d4bc
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─USB2.0 Hub:
      Device ID:          b0a78eb71f4eeea7df8fb114522556ba8ce22074
      Summary:            USB 2.x Hub
      Current version:    9.153
      Vendor:             VIA Labs, Inc. (USB:0x2109)
      Install Duration:   15 seconds
      GUIDs:              c3a147e8-7d29-5669-bed2-f60cecdd2cf2
                          20054c2d-afa4-5f73-ade0-18dbd6f09fa5
                          ae960a48-da94-5862-94b0-f38f69e4f74d
                          f3397d51-d9c9-5306-8992-362bc3c56a71
                          2546d7af-3b17-5174-8ff5-e92f441a926f
                          b0a54e37-45da-53f3-98c6-4b33a105d30e
      Device Flags:       • Updatable
                          • Cryptographic hash verification is available
                          • Device stages updates
                          • Device can recover flash failures
    
Idle…                    [***************************************]
Successfully uploaded 1 report
```