fwupd/firmware-dell #56: Latitude 3490/3590 missing completely from list (as best I can tell) including no BIOS

pjwelsh, 2021-04-06:
The Dell Latitude 3490/3590 BIOS listing seems to be entirely missing from the Dell Latitude list. There are older 34x0 and 35x0 models listed, however.
Please add an entry for the Latitude 3490 along with latest BIOS.

Latest "Urgent" Dell Latitude 3490 and 3590 System BIOS is at:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=6dfrt&oscode=biosa&productcode=latitude-14-3490-laptop

superm1, 2021-04-11:
@therealjuanmartinez can you raise this internally.

pjwelsh, 2021-05-11:
Poke: Any headway getting this completely missing setup added to fwupd?

pjwelsh, 2021-05-25:
The "@therealjuanmartinez" doesn't seem to have fwupd listed as a repository associated with them. Are they the correct person?
Any Dell person able to help move this forward? Thanks!

superm1, 2021-05-25:
@dell-client-linux can you look?

dell-client-linux, 2021-05-27:
Ok, we are working on this issue now, please wait.

pjwelsh, 2021-06-02:
Poke for missing BIOS update.

pjwelsh, 2021-06-09:
Poke for missing completely missing system and it's Urgent BIOS updates.

pjwelsh, 2021-06-30:
Getting close to 3 months now Dell :(

dell-client-linux, 2021-07-05:
These two systems are not supported for fwupd on LVFS currently. The short term solution is to use the BIOS in Dell Support website to update in BIOS boot menu. We are still trying to make these two platforms in the fwupd scope. Sorry for the inconvenience.

pjwelsh, 2021-09-01:
Any progress on adding these two systems (Latitude 3490/3590)?

pjwelsh, 2021-10-13:
Not a complaint, but is there more information as why the 3490/3590 Latitude is not in "scope"? I have several systems that fall into this category. The older 3480 is in scope as well as the newer 3410.
As an admin to a fair number of Dell systems (servers, laptops and desktops/workstations - all running Linux), I'm looking to (1) understand the situation, (2) adjust our inventory to reflect such issues and (3) inform all to the importance or maintaining Linux friendly components by a company that buys products.

pjwelsh, 2022-01-31:
I'm still staying tuned... Any progress on adding this longgggg missing BIOS?

pjwelsh, 2022-02-20:
Any progress on getting these added to lvfs?

pjwelsh, 2022-03-10:
@dell-client-linux Any progress on getting these added to lvfs?

pjwelsh, 2022-04-28:
Here is a repeat of my comment on the "scope" statement from last year:
"Not a complaint, but is there more information as why the 3490/3590 Latitude is not in "scope"? I have several systems that fall into this category. The older 3480 is in scope as well as the newer 3410."
How did the before and after version of the laptop get in scope and this one not in scope? I've had this ticket open for > 1 year now.

sunnycrown83, 2022-04-28:
The context here is LVFS was starting to support on Dell's hardware about 3-4 years ago. It didn't support all of the platforms at beginning but expanded to more and more platforms incrementally based on internal process. I'm sorry it's hard to go back to ask for those platforms that didn't include LVFS at beginning but still I'm checking with internal team that if there is opportunity to promote the FW. Sorry again about the inconvenience.

sunnycrown83, 2022-05-10:
@pjwelsh Could you help have a double check？ Latitude 3X90 00.1.19.0 should have been added.
Thanks.

pjwelsh, 2022-05-10:
Success! The 3490 I tested with both downloaded and went through the BIOS update process with no issues.
Thank you very much for your assistance in pushing this into the fwupd realm!
closing.

closed 2022-05-10 by pjwelsh