fwupd/firmware-dell #57: Precision 5750 firmware outdated

adrien-dorsaz, 2021-04-15:
Hello,

I've seen some updates were available for Dell Precision 5750 laptop.

The latest UEFI from DELL support is [1.7.2](https://www.dell.com/support/home/fr-ch/drivers/driversdetails?driverid=tf09f&oscode=biosa&productcode=precision-17-5750-laptop) currently.

On LVFS the [1.4.0](https://fwupd.org/lvfs/devices/com.dell.uefi7d606caf.firmware) version is available..

spockfish, 2021-05-04:
This is the same firmware as is being used for the XPS 9700 / 9500.

dell-client-linux, 2021-05-28:
@adrien-dorsaz We have updated the version on LVFS, now is 1.7.2.  You can check it.

adrien-dorsaz, 2021-06-01:
Thanks for the notification. I've downloaded the file from your link and tried to install it.

Unfortunately, it didn't work on my Precision 5750 and Debian Bullseye:

```
$ sudo fwupdmgr install --verbose 470763daf8a35123f8ad8501879695f76311df1d4706e7ce94392ff9b1916306-XPS_17_9700_Precision_5750_1.7.2.cab
(fwupdmgr:11411): GLib-DEBUG: 13:08:53.539: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:11411): Fwupd-DEBUG: 13:08:53.553: Emitting ::status-changed() [idle]
(fwupdmgr:11411): Fwupd-DEBUG: 13:08:53.572: Emitting ::status-changed() [decompressing]
Decompressing…           [ -                                     ](fwupdmgr:11411): Fwupd-DEBUG: 13:08:54.324: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
No supported devices found
```
I don't know how to help, but see below the list of devices found on my machine:

```javascript
$ fwupdagent get-devices
{
  "Devices" : [
    {
      "Name" : "Thunderbolt host controller",
      "DeviceId" : "ac32bb6e3ca9389f845bda7674fe9b772735e287",
      "Guid" : [
        "580f0596-7dc9-5af1-a401-effa8a2dd091",
        "44e42752-0136-5e98-812e-7a7e2c876b3e",
        "cd8d2b1e-ca51-5649-87ad-d3cbe54483c6",
        "b690ada5-1737-5838-8992-044b771e92b9"
      ],
      "Summary" : "Unmatched performance for high-speed I/O",
      "Plugin" : "thunderbolt",
      "Protocol" : "com.intel.thunderbolt",
      "Flags" : [
        "internal",
        "updatable",
        "require-ac",
        "registered",
        "dual-image"
      ],
      "Vendor" : "Dell",
      "VendorId" : "THUNDERBOLT:0x00D4|TBT:0x00D4",
      "VendorIds" : [
        "THUNDERBOLT:0x00D4",
        "TBT:0x00D4"
      ],
      "Version" : "56.00",
      "VersionFormat" : "pair",
      "Icons" : [
        "thunderbolt"
      ],
      "Created" : 1622527894
    },
    {
      "Name" : "DELL0990:00 04F3:311C",
      "DeviceId" : "b8cfad8dcc60878190ea039a952b3d7cd97cad91",
      "Guid" : [
        "9b6fef1a-77aa-565c-9983-9da2dd2c932b",
        "46598f8d-5777-50c4-8da9-cda0cc8bbc26",
        "d00322f0-ff1f-5458-9c64-8a771fb0b72e",
        "4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96",
        "6cf7ae6a-8b98-5b63-9af8-b23b932f65de"
      ],
      "Summary" : "Elan Touchpad",
      "Plugin" : "elantp",
      "Protocol" : "tw.com.emc.elantp",
      "Flags" : [
        "internal",
        "updatable",
        "registered"
      ],
      "VendorId" : "HIDRAW:0x04F3",
[... 732 lines cut ...]
      "Guid" : [
        "90b001b4-a822-5230-8f3b-bc38ae41afd3",
        "f248f28f-594f-55af-bfe9-96d3373cf8dc",
        "6c3dbf6c-4e6f-5309-9954-c5ab7aca617e",
        "3777783a-3f83-56a5-95f4-533eb6a2bd19"
      ],
      "Plugin" : "optionrom",
      "Flags" : [
        "internal",
        "registered",
        "can-verify",
        "can-verify-image"
      ],
      "Vendor" : "Intel Corporation",
      "VendorId" : "PCI:0x8086",
      "Version" : "05",
      "VersionFormat" : "plain",
      "Icons" : [
        "audio-card"
      ],
      "Created" : 1622527892
    }
  ]
}
```

adrien-dorsaz, 2021-06-15:
I had some holidays and I'm back today.

I've applied some Debian Bullseye updates and then I was able to install this firmware after restart of the fwupd service and run `fwupdmgr refresh --verbose --force`.

Thanks :+1:

closed 2021-06-15 by hughsie