fwupd/firmware-dell #58: Stake firmware for Micron 2200S SSD (in Latitude 5501)

rzerda, 2021-04-28:
Hello,

Micron 2200S SSD has version 2200.1060 on the Dell site, but 2200.1050 on fwupd.org:

* Dell: https://www.dell.com/support/home/ru-ru/drivers/driversdetails?driverid=9w77n&oscode=wt64a&productcode=latitude-15-5501-laptop
* fwupd.org: https://fwupd.org/lvfs/devices/com.dell.108489.firmware

LaHaine, 2022-05-04:
The current Dell firmware for the Micron 2200S is 1070:
https://www.dell.com/support/home/de-de/drivers/driversdetails?driverid=3jg3g&oscode=wt64a&productcode=optiplex-7060-desktop
Still nothing newer than 1050 on fwupd.org.

CragW, 2022-05-04:
The firmware has just been released, @rzerda and @LaHaine can you try again? Thanks.

LaHaine, 2022-05-04:
I think I'll have to try later:
```
# fwupdmgr refresh --force
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```
This is on Ubuntu 18.04 LTS.

CragW, 2022-05-04:
I think you're hitting this issue fwupd/fwupd#391.

LaHaine, 2022-05-04:
I could fetch and install the firmware update, but I cannot reboot now, will have to do that later.