fwupd/firmware-dell #59: Dell Inspiron 3467 outdated and 3567 missing

haunt98, 2021-05-04:
Dell support link: https://www.dell.com/support/home/en-vn/drivers/driversdetails?driverid=c719v&oscode=wt64a&productcode=inspiron-15-3567-laptop

Both 3467 and 3567 use the same firmware with v2.14.0

LVFS link:  https://fwupd.org/lvfs/devices/com.dell.uefi1d4362ca.firmware

I have only found 3467 on LVFS but it is outdated (v01.05.00).

And 3567 is missing on LVFS

pjwelsh, 2022-08-22:
@dell-client-linux This system seems to have been overlooked for a very long time. The BIOS available is 1.5.0 from 2017 (https://fwupd.org/lvfs/devices/com.dell.uefi1d4362ca.firmware) while the most recent from Dell this month is 2.20.0 (https://www.dell.com/support/home/en-vn/drivers/driversdetails?driverid=hn9fg&oscode=wt64a&productcode=inspiron-15-3567-laptop).
Any chance to update, please.

pjwelsh, 2022-09-01:
@dell-client-linux Looks like this has been updated to the 2.20.0 BIOS version... silently... Thank you to whomever did it, but please let us know (via the open issues).
One small gripe is a typo in "*System BIOS 2.2.0 for Inspiron 3467/3567 system". Note it should be "2.20.0" not "2.2.0".
@haunt98 are you able to verify this does what you expect now and close the ticket accordingly or comment with issue?

closed 2022-09-02 by haunt98