fwupd/firmware-dell #60: fwupdmgr tries to install wrong firmware on Dell Latitude 7390 2-in-1

tevrugt, 2021-05-05:
**Describe the bug**
`sudo fwupdmgr update` tries to install firmware  "Firmware for the Dell Latitude E5X60" on a Dell Latitude 7390 2-in-1.

**Steps to Reproduce**
`sudo fwupdmgr get-updates` on a Dell Latitude 7390 2-in-1 lists firmware of the Dell Latitude E5X60:
```
Latitude 7390 2-in-1 System Firmware has firmware updates:
Device ID:               4a91463cbd349e9902ad7c61fcb57e4b41119b9e
GUID:                    6180aaaa-5529-4bbf-b4fd-65b4f788de5b
ID:                      com.dell.uefi6180aaaa.firmware
Update Version:          0.1.25.3
Update Name:             Latitude E5X60 System Update Update
Update Summary:          Firmware for the Dell Latitude E5X60
Update Remote ID:        lvfs
Update Checksum:         SHA1(a05a86d43441ae135c0d4eae69a26b719ab9d794)
Update Location:         https://fwupd.org/downloads/f58bd3dd5626827bc80db6970cb7235a315c3473a6c6e095e4bf138da28e6e79-Latitude_E5x70_Precision_3510_1.25.3.cab
Update Description:      This stable release fixes the following issues:
                         
                         Some new functionality has also been added:                    
[...]
```

`sudo fwupdmgr update` results in:          
```
Downloading 0.1.25.3 for Latitude 7390 2-in-1 System Firmware...
Fetching firmware https://fwupd.org/downloads/f58bd3dd5626827bc80db6970cb7235a315c3473a6c6e095e4bf138da28e6e79-Latitude_E5x70_Precision_3510_1.25.3.cab
```
and the laptop attempts to update the laptop with the wrong bios on next reboot (without success)

Besides the wrong firmware for the E5X60 (latest version: 0.1.25.3), LVFS also lists the correct firmware for the 7390 2-in-1 (latest version 0.1.18.0):
[https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.firmware](https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.firmware)

Today I downloaded v0.1.18.0 and successfully installed this firmware manually (`sudo fwupdmgr install ./4b59ee9f03ade4753ce5c54d54e922418d41e5ad18a524677e087bdb93068420-Latitude_7390_2in1_1.18.0.cab`).

**Expected behavior**
Only firmware for the Dell Latitude 7390 2-in-1 should be retrieved and installed on this machine.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
```
client version:	1.2.13
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.13
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`apt`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
```
Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.83.3874
  VersionBootloader:    11.8.83.3874
  VersionFormat:        intel-me
  Icon:                 computer
  Created:              2021-05-05

Latitude 7390 2-in-1 System Firmware
  DeviceId:             4a91463cbd349e9902ad7c61fcb57e4b41119b9e
  Guid:                 6180aaaa-5529-4bbf-b4fd-65b4f788de5b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(bf833c47302480fcc2a7ed7ba98e85e7af5a8f7c)
  Checksum:             SHA256(c31a2074171bc7c1df1db2b9ba0be22c435b73f011c04b62e7e0dc54fe2d3fe7)
  Vendor:               Dell Inc.
  Version:              0.1.18.0
  VersionLowest:        0.1.18.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2021-05-05
  UpdateState:          success

Latitude 7390 2-in-1 TPM 2.0
  DeviceId:             c3319ca6284cd7c9d002c2649c7733528aae35d7
  Guid:                 22ade82f-8293-550a-a90e-5cd7c4541c71
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Version:              7.2.0.1
  VersionFormat:        quad
  Icon:                 computer
  Created:              2021-05-05

UHD Graphics 620
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 fed2efa4-7045-55a1-b4fc-b29283d59fe5
  Guid:                 8b72a10c-1279-5f8e-a28a-34fb11a58240
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Version:              07
  VersionFormat:        plain
  Icon:                 audio-card
  Created:              2021-05-05

PC401 NVMe SK hynix 1TB
  DeviceId:             6cc1c2d37a98d92020a9c80f9cc15d88a0656384
  Guid:                 2260c743-963b-566e-93f1-8223c3e6b87b
  Guid:                 36f86f0d-aede-5c78-8273-6b4132637313
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               SK hynix
  VendorId:             NVME:0x1C5C
  Version:              80003E00
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2021-05-05
```

**Additional questions**
- Operating system and version: Debian 10, updated May-5 (`apt update`, `apt upgrade`, `apt full-upgrade`)
- Have you tried rebooting? Yes
- Is this a regression? Not to my knowledge.

hughsie, 2021-05-05:
@dell-client-linux can you have a look at this as a priority please. Thanks!

tevrugt, 2021-05-05:
Missed to refer to the existing bugs, sorry:
[Dell Latitude 7390 2-in-1 shows firmware updates for Latitude E7x60 #770](https://github.com/fwupd/fwupd/issues/770)
[fwupd proposing incorrect firmware for Dell Latitude 7390. #695](https://github.com/fwupd/fwupd/issues/695)

dell-client-linux, 2021-05-06:
Will check internally to see if anything wrong for the ID.

superm1, 2021-05-06:
I think proper solution is most likely to add CHID.

CragW, 2021-05-06:
For E5X60 firmware incorrectly listed up, issue should be fixed in Latitude 7x90 rev. 1.19.0.
https://fwupd.org/lvfs/devices/com.dell.uefia86a3f07.firmware

tevrugt, 2021-05-06:
Thanks for the quick response! I'll check the behavior of `fwupdmgr` a.s.a.p., I'm currently at another machine.

A browser-based observation:
- Searching for "Dell Latitude 7390 2-in-1" in the search field on [https://fwupd.org/](https://fwupd.org/) leads to [https://fwupd.org/lvfs/search?value=Dell+Latitude+7390+2-in-1](https://fwupd.org/lvfs/search?value=Dell+Latitude+7390+2-in-1).
- The link [Latitude 7390 2-in-1 System Update](https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.firmware) directs to [Dell Latitude E5X60](https://fwupd.org/lvfs/devices/com.dell.uefi6180aaaa.firmware), so the 7390 2-in-1 and E5X60 still seem to be mixed up somehow.

CragW, 2021-05-07:
The GUID differentiation isn't set for "Dell Latitude 7390 2-in-1", let's check into @superm1's suggested CHID. I will hide previous comment that was pointing to "Dell Latitude 7x90" to avoid confusion.

tevrugt, 2021-05-08:
> Thanks for the quick response! I'll check the behavior of `fwupdmgr` a.s.a.p., I'm currently at another machine.
>
> A browser-based observation:
>
>     * Searching for "Dell Latitude 7390 2-in-1" in the [...]

Addition, confirms observation from web-page:

`$ sudo fwupdmgr refresh` and `$ sudo fwupdmgr get-updates | grep -E "ID|Name"` results in

```
Device ID:               4a91463cbd349e9902ad7c61fcb57e4b41119b9e
GUID:                    6180aaaa-5529-4bbf-b4fd-65b4f788de5b
ID:                      com.dell.uefi6180aaaa.firmware
Update Name:             Latitude E5X60 System Update Update
Update Remote ID:        lvfs
ID:                      com.dell.uefi6180aaaa.firmware
Update Name:             Latitude E5X60 System Update Update
Update Remote ID:        lvfs
ID:                      com.dell.uefi6180aaaa.firmware
Update Name:             Latitude E5X60 System Update Update
Update Remote ID:        lvfs
ID:                      com.dell.uefi6180aaaa.firmware
Update Name:             Latitude E5X60 System Update Update
Update Remote ID:        lvfs
ID:                      com.dell.uefi6180aaaa.firmware
Update Name:             Latitude E5X60 System Update Update
Update Remote ID:        lvfs
```

`$ sudo fwupdmgr update`:

```
Downloading 0.1.25.3 for Latitude 7390 2-in-1 System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
```

superm1, 2021-05-08:
I would suggest to treat the web UI and command line tools separately. Web UI is only showing based on GUID. I think it needs some work to handle this case of two systems sharing guid and what to render.