fwupd/firmware-dell #63: Urgent OptiPlex 7040 1.19.0 BIOS missing

pjwelsh, 2021-05-12:
LVFS latest page 1.18.1 https://fwupd.org/lvfs/devices/com.dell.uefib950d63c.firmware
Dell latest 1.19.0 https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=cc44x&oscode=biosa&productcode=optiplex-7040-desktop
OptiPlex 7040

pjwelsh, 2021-06-02:
Poke for missing BIOS update.

pjwelsh, 2021-06-09:
Poke for missing "Urgent" BIOS update.

pjwelsh, 2021-06-09:
This is odd, the web site shows the 1.18.1 version is latest version, however, the one fwupmgr installed is the correct newer 1.19.0.
Not sure if the version update *just* happened and web page has not yet caught up or what....

pjwelsh, 2021-06-09:
Sorry, just deleted an extraneous comment as I found out the system had been updated via the install_dell_bios_exe.py tool in this case.

pjwelsh, 2021-06-30:
Poke for missing "Urgent" BIOS update notice from 1.5 months ago.

pjwelsh, 2021-07-06:
Appears to have silently been updated. closing.

closed 2021-07-06 by pjwelsh

hughsie, 2021-07-06:
@pjwelsh thanks for keeping track of these; I appreciate this.