fwupd/firmware-dell #64: Urgent OptiPlex 7070 1.7.2 BIOS missing

pjwelsh, 2021-05-12:
LVFS 1.7.0 https://fwupd.org/lvfs/devices/com.dell.uefiaecf8a53.firmware
Dell latest 1.7.2 https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=kygvt&oscode=biosa&productcode=optiplex-7070-desktop
OptiPlex 7070

andre195, 2021-05-30:
Hi folks, checking the Dell site I noticed this as well:

Missing BIOS Firmware: OptiPlex 7070 Version 1.7.2

A link to the firmware update at Dell support
[https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=kygvt](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=kygvt)

A link to the previous firmware upgrade at LVFS
[https://fwupd.org/downloads/c09a8f940b54e62f9de0483d956d91fd38a9f3bf14fa890f9519bae41d09badc-Dell_OptiPlex_7070_BIOS_1.7.0.cab](https://fwupd.org/downloads/c09a8f940b54e62f9de0483d956d91fd38a9f3bf14fa890f9519bae41d09badc-Dell_OptiPlex_7070_BIOS_1.7.0.cab)

The system
Dell OptiPlex 7070 Desktop

Thanks!

pjwelsh, 2021-06-02:
Poke for missing BIOS update.

pjwelsh, 2021-06-09:
Poke for missing "Urgent" BIOS update.

pjwelsh, 2021-06-30:
Poke for missing "Urgent" BIOS update notice from 1.5 months ago.

pjwelsh, 2021-07-06:
Appears to have silently been updated. closing.

closed 2021-07-06 by pjwelsh

andre195, 2021-07-10:
I'm still seeing 1.7.0 here [https://fwupd.org/lvfs/search?value=OptiPlex+7070](https://fwupd.org/lvfs/search?value=OptiPlex+7070) am I doing something wrong?

pjwelsh, 2021-07-10:
It is in the lvfs-testing  "remote. Not sure why it was put/left there,
however.
fwupdmgr get-remotes
...
Remote ID:               lvfs-testing
Title:                   Linux Vendor Firmware Service (testing)
Type:                    download
Keyring:                 gpg
Enabled:                 true
...

Previously I ran a:
fwupdmgr enable-remote lvfs-testing
to get that to be enabled.
PJ

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived.  Cassens