fwupd/firmware-dell #68: Dell Latitude 7480 wrong version number

zoltankiss-ericsson, 2021-05-28:
**Describe the bug**
fwupd claims that there is a new UEFI for my laptop with version number 1.20.2. However Dell's support page doesn't show this version, but only 1.20.1. Comparing the version history, they are the same, so I guess the numbers were mixed up. "fwupdmgr get-releases" shows there is no 1.20.1, but only 1.20.2. 
This issue tracker is the nearest thing I've found to contact anyone, I hope someone can point to the right direction

**Steps to Reproduce**
"fwupdmgr get-updates" shows there is an update, while there shouldn't be really.

**Expected behavior**
No update shown

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt

**I'm skipping the next 2 as not relevant**
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**System UEFI configuration**
Please provide the output of the following commands:
```shell
$ efibootmgr -v
```

```shell
$ efivar -l | grep fw
```

```shell
$ tree /boot
```

**Additional questions**
- Operating system and version: Ubuntu 18.04
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

dell-client-linux, 2021-07-05:
It's the same version between LVFS and Dell support to fix the same CVE. Should be typo when version submitted. Will correct it in next version.

pjwelsh, 2022-02-20:
This issue should be closed by someone who can. OP reports, "...managed to upgrade successfully with fwupd...".

closed 2022-02-20 by hughsie