fwupd/firmware-dell #69: Dell XPS 15 9550 not detected

Gelmo, 2021-05-28:
**Describe the bug**
This issue is the same that was described in issue fwupd/fwupd#2671 

Just wanted to report that I had the same issue today with the same laptop on 1.13.1 (Device ID 2da31270d317b076424992de14a0f08ae373c137 and GUID 124c207d-5db8-4d95-bd31-34fd971b34f9). There still seems to be something interfering that you may want to review for other device owners.

fwupdmgr did not find this as a supported device, even after a forced refresh. I manually installed 1.14.0 via fwupdtool.

After the update to 1.14.0, fwupdmgr still does not see this as a supported device. I submitted a report via get-devices a few hours ago, in case that's helpful.

The Dell site suggests that the latest bios for this is 1.14.0. When searching the GUID on fwupd.org, the results show that the latest version is 1.16.1. Should I attempt to manually install 1.16.1? Is that version not intended for my device? If so, perhaps the listings at fwupd.org need more adjustments. Let me know if there's anything I could provide to assist with this.

**General Info**
```shell
$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
```

```shell
$ sudo fwupdmgr get-releases
No supported devices
```

**Expected behavior**
The device should be detected and supported, as the GUID is listed at fwupd.org.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.8
```

Installed via Pop OS's default configuration

**fwupd device information**
```shell
$ sudo fwupdmgr get-devices --show-all-devices
XPS 15 9550
│
├─BCM920703 Bluetooth 4.1:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Current version:    1.12
│     Vendor:             Broadcom Corp (USB:0x0A5C)
│     Serial Number:      441CA8E1EC60
│     GUIDs:              35fd074d-7c78-599e-b37a-88931c63ba01 ← USB\VID_0A5C&PID_6410&REV_0112
│                         f8498646-6724-5f55-9806-57b3f8636ad8 ← USB\VID_0A5C&PID_6410
│     Device Flags:       • Updatable
│
├─GM107M [GeForce GTX 960M] (XPS 15 9550):
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a2
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              5d36a05e-42ad-5b87-8a2d-27b8bd7bfe04 ← PCI\VEN_10DE&DEV_139B&SUBSYS_102806E4&REV_A2
│                         4bbd4dbf-83dd-58ea-a5ad-081ca6f8bca9 ← PCI\VEN_10DE&DEV_139B&SUBSYS_102806E4
│                         995dfa05-5863-56db-8087-221c0cce69a4 ← PCI\VEN_10DE&DEV_139B&REV_A2
│                         2195ad44-2598-565c-819e-16026a8d2f22 ← PCI\VEN_10DE&DEV_139B
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─HD Graphics 530 (XPS 15 9550):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              bf3a0b36-1995-5fe3-b81d-9effc018f659 ← PCI\VEN_8086&DEV_191B&SUBSYS_102806E4&REV_06
│                         cc0161dd-f85a-5d3c-8a6f-329aeffd9c55 ← PCI\VEN_8086&DEV_191B&SUBSYS_102806E4
[... 151 lines cut ...]
│       │   └── Recovery-DA8B-2159.conf
│       ├── loader.conf
│       └── random-seed
├── initrd.img -> initrd.img-5.11.0-7614-generic
├── initrd.img-5.11.0-7612-generic
├── initrd.img-5.11.0-7614-generic
├── initrd.img-5.3.0-7642-generic
├── initrd.img.old -> initrd.img-5.11.0-7612-generic
├── System.map-5.11.0-7612-generic
├── System.map-5.11.0-7614-generic
├── vmlinuz -> vmlinuz-5.11.0-7614-generic
├── vmlinuz-5.11.0-7612-generic
├── vmlinuz-5.11.0-7614-generic
└── vmlinuz.old -> vmlinuz-5.11.0-7612-generic

16 directories, 34 files
```

**Additional questions**
- Operating system and version: Pop OS (tested on 20.04 and 20.10)
- Have you tried rebooting? Yes.
- Is this a regression? Perhaps; issue fwupd/fwupd#2671 suggests that this was previously corrected.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

GoPerry, 2021-06-01:
> Since the BIOS cab is built by automation and presumably the XPS version isn't supposed to have more updates it might be an easier process to rename the ID on all the old ones that apply to both to me.

Mario.
So 

> OK I think I found what's going on. If you look at the stable metadata, you can see that
> 
> ```
>     <requires>
>       <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
>     </requires>
> ```
> 
> Is set for the device in `firmware.xml.gz` _not_ in the **release**. So it is applied to _all_ releases.
> 
> ```
>       <release version="68354" timestamp="1564495178" urgency="critical">
>         <location>https://fwupd.org/downloads/8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab</location>
>         <checksum type="sha1" filename="8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab" target="container">767bec5099a9ce9278ee95860fd64018af1617d5</checksum>
>         <checksum type="sha256" filename="8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab" target="container">257a690baf81eec1ef98e5e57c31aede3055e5f6547c22629f99795d5ebba6d3</checksum>
>         <checksum type="sha1" filename="firmware.bin" target="content">8e335187c2946574c103b0e44dc5d6747416e0ac</checksum>
>         <checksum type="sha256" filename="firmware.bin" target="content">3e571e0b81239b23f33f59717541c447aa794bbea0c32f656c8b688fd5422463</checksum>
>         <description>
>           <p>This stable release fixes the following issues:</p>
>           <ul>
>             <li>Fixed the issue where the TPM option disappears from the BIOS setup menu intermittently.</li>
>           </ul>
>         </description>
>         <size type="installed">10240280</size>
>         <size type="download">10260780</size>
>         <artifacts>
>           <artifact type="binary">
>             <location>https://fwupd.org/downloads/8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab</location>
>             <filename>Precision_5510_1.11.2.cab</filename>
>             <checksum type="sha1">767bec5099a9ce9278ee95860fd64018af1617d5</checksum>
>             <checksum type="sha256">257a690baf81eec1ef98e5e57c31aede3055e5f6547c22629f99795d5ebba6d3</checksum>
>             <size type="installed">10240280</size>
>             <size type="download">10260780</size>
>           </artifact>
>         </artifacts>
>       </release>
>     </releases>
>     <requires>
>       <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
>     </requires>
>   </component>
> ```

Hi Mario.
So the fix will be that move the <hardware> section to release section like this ?
Is it correct ?

      <release version="68354" timestamp="1564495178" urgency="critical">
      .....skip....
   ++ <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </releases>

    <requires>
   -- <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>

GoPerry, 2021-06-01:
Hi Mario.
So the fix will be that move the <hardware> section to release section like this ?
Is it correct ?

      <release version="68354" timestamp="1564495178" urgency="critical">
      .....skip....
       ++ <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </releases>

    <requires>
   -- <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>

superm1, 2021-06-01:
@GoPerry the fix is is to change the appstream ID, ie what is going into the top as `com.dell.uefiXXXXXX`

superm1, 2021-06-01:
Because it's built by automation on the Dell side - I would suggest change appstream ID on all the older ones to something common instead.

GoPerry, 2021-06-03:
> @GoPerry the fix is is to change the appstream ID, ie what is going into the top as `com.dell.uefiXXXXXX`

Hi Mario.
Because I have no access the to automation build server, I would need to check with team
It is clear what we should fix now.

Thanks  a lot !

superm1, 2021-06-03:
At least my experience with how that works I think it's easier to just fixup all old releases on lvfs than changing build server, but up to you which way to go!

JoeyCorleone, 2022-07-12:
After refreshing the metadata, fwupmgr now suggests a BIOS update:

`Upgrade System Firmware from 1.14.0 to 1.22.0?`

However, on the DELL website, 1.14 is still the [latest available version](https://www.dell.com/support/home/en-us/product-support/product/xps-15-9550-laptop/drivers) for the XPS 9550.

In the get-updates summary, both XPS 15 9550 and Precision 5510 are mentioned:

`├─XPS 15 9550/Precision 5510:
  │     New version:      1.22.0
  │     Remote ID:        lvfs
  │     Release ID:       14400
  │     Summary:          Firmware for the Dell XPS 15 9550/Precision 5510
  │     License:          Proprietary
  │     Size:             13,4 MB
  │     Created:          2022-06-06
  │     Urgency:          High
  │     Vendor:           Dell
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     This stable release fixes the following issues:
  │     
  │     • Improved the system stability.
`

Is it correct to assume that the update is safe because it is offered by fwupdmgr?

chinyakao, 2022-07-22:
Hi @JoeyCorleone,

XPS 15 9550, Berlinetta was EOS on 2020/3/31. And due to Berlinetta P, and Berlinetta use same BIOS package, when we submit the latest BIOS package to LVFS, it also show the info of xml. 
We have remove the tag of XPS 15 9550 from the latest BIOS 1.22.0.

Thank you

chinyakao, 2022-07-22:
Hi @JoeyCorleone 
They have fixed the filename for Precision 5510 release to avoid confusion.

closed 2022-08-23 by CragW