fwupd/firmware-dell #70: XPS 9360 NVM firmware update missing

pachulo, 2021-05-31:
Link to firmware update at OEM website:
* https://www.dell.com/support/home/es-es/drivers/driversdetails?driverid=mp2n1&oscode=wt64a&productcode=xps-13-9360-laptop

Firmware release date (if available on OEM website)
2021-05-21

PhilArmstrong, 2022-05-23:
This has been superceded by another Intel Thunderbolt update dated 5th November 2021:

  https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=tbtc0&oscode=wt64a&productcode=xps-13-9360-laptop

Is there any chance of poking Dell into putting the firmware files into the LVFS, or are we doomed? Dell are putting BIOS updates in to the LVFS for this laptop, but apparently not Thunderbolt updates!

PhilArmstrong, 2022-05-23:
Reading the CVEs It’s possible that both of the above updates are to fix issues in the Windows device however. Perhaps the actual Thunderbolt firmware is unchanged? Unfortunately Intel have changed how they package the firmware updates in the exe that Dell supplies & I can’t work out how to unpack the current releases - they don’t seem to be using any of the standard windows packing formats like .cab or 7zip.

CragW, 2022-05-24:
Given link the mentioned release is Thunderbolt driver stack in Windows, for firmware please visit https://www.dell.com/support/home/en-uk/drivers/driversdetails?driverid=d1x3v simarily https://fwupd.org/lvfs/devices/com.dell.tbtdfd51125.firmware on LVFS.

closed 2022-05-24 by CragW

PhilArmstrong, 2022-05-24:
I guess Intel bundled firmware updates into the Windows binary releases in the past, hence the confusion? Thanks for checking anyway.

pachulo, 2022-05-25:
I'm not seeing any update for my `Thunderbolt host controller`:
```
$ sudo fwupdmgr upgrade 
Devices with no available firmware updates: 
 • Dell TB16 EC
 • Dell TB16 Port Controller 1
 • Dell TB16 Port Controller 2
 • Dell TB16 Thunderbolt Cable
 • THNSN5512GPUK NVMe TOSHIBA 512GB
 • Thunderbolt Cable
 • Thunderbolt Dock
 • UEFI dbx
Devices with the latest available firmware version:
 • System Firmware
 • Thunderbolt host controller
```

is that expected @CragW ?

By the way, I do see this now with `get-devices`:
```
$ fwupdmgr get-devices
...
─Thunderbolt host controller:
      Device ID:          19e719f0ad60e646945ff192ba159fa70bc1cda2
      Summary:            Unmatched performance for high-speed I/O
      Current version:    26.01
      Vendor:             Dell (THUNDERBOLT:0x00D4, TBT:0x00D4)
      GUID:               0c87187d-fc4e-5177-915f-61dbb4160aaf
                          99b7b115-1d63-5cee-b687-bd841ff19996
                          dfd51125-338f-56ff-b721-fa3bea8e534e
                          767e33a6-8a8f-5b11-bc1a-aa881e5e2bb1
```

CragW, 2022-05-26:
@pachulo, you have the latset thunderbolt firmware 26.01 installed. 👍