fwupd/firmware-dell #75: Dell 4 Zone Keyboard V00.15A

GMMan, 2021-07-02:
As requested [here](https://twitter.com/hughsient/status/1410875533575995394?s=20)

Driver: https://www.dell.com/support/home/en-ca/drivers/driversdetails?driverid=h6kn2  
System: Alienware m15/m17 R3/R4  
No previous version found on LVFS

Note the driver package is very unorthodox per Twitter thread.

hughsie, 2021-07-02:
@dell-client-linux can you share what update protocol this is using?

dell-client-linux, 2021-07-05:
@hughsie, this keyboard and the Alienware systems are not supported with Linux in Dell. Therefore, there is no FW update for Linux  right now.

hughsie, 2021-07-05:
@dell-client-linux I know we don't support the Alienware systems, but I wondered if we could work on the keyboard so that it would qualify for the _Works With ChromeBook_ requirements. If you think it's a possibility, I'd be happy if you could send an email to the peripherals group and I can make a case for adding support.