fwupd/firmware-dell #78: OptiPlex 7080 missing Urgent BIOS update 1.4.1

pjwelsh, 2021-07-28:
LVFS has https://fwupd.org/lvfs/search?value=OptiPlex+7080
Dell has 1.4.1 https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=cccj7&oscode=biosa&productcode=optiplex-7080-desktop

OptiPlex 7080 Urgent BIOS update for several CVEs released ~7/23/21:
Fixes & Enhancements
- Firmware updates to address the Intel Security Advisory INTEL-SA-00463 (CVE-2020-12359).
- Firmware updates to address the Intel Security Advisory INTEL-SA-00464 (CVE-2020-24511).
- Firmware updates to address the Intel Security Advisory INTEL-SA-00459 (CVE-2020-24507, CVE-2020-8703, CVE-2020-24506, and CVE-2020-24516).
- Firmware updates to address the Intel Security Advisory INTEL-SA-00463 (CVE-2020-8700).
- Firmware updates to address security vulnerabilities.
- Firmware update to address CVE-2020-26187.
- Firmware security updates to address CVE-2021-21571 and CVE-2021-21572.

pjwelsh, 2021-08-10:
Update to 1.4.1 is posted it seems.
Not sure if the person/people actually doing the updates are on this list. I keep having BIOS updates added, but not any comments on my tickets.
Any chance to validate that the *correct* people as seeing these types of requests? I get the feeling they are not.

hughsie, 2021-08-10:
@pjwelsh I don;t think the people uploading firmware have permission to close these issues. I'm pretty sure @dell-client-linux is shouting at the right people when required :)

closed 2021-08-10 by hughsie

pjwelsh, 2021-08-10:
Closing is not the issue... it's the lack of comments.
Thank you for responding. This is very much appreciated.