fwupd/firmware-dell #82: Dell XPS 13 7390 BIOS update 1.9.0 breaks laptop

mercuriete, 2021-08-19:
After a BIOS update i git a kernel error and prompts me to initramfs due to failure on mounting root partion.

Output:

```
[drm: intel_bios_init  [i915]] *ERROR* Unexpected child device config size 39 (expected 38 for VBT version 228)
```

I will try to flash a previous firmware but I dont know how to do It.

Thanks.

mercuriete, 2021-08-19:
[image]

This is the last image I could take with my laptop working.
I don't even know why I take that photo in the first place.

maybe I already know what was going to happen. :(

mercuriete, 2021-08-19:
[image]

This is the output from the kernel while trying to mount root partition.

mercuriete, 2021-08-19:
After rolling back to 1.8.0 using this tutorial:
https://www.dell.com/support/kbdoc/es-es/000128928/actualizacion-del-bios-desde-el-menu-de-arranque-unico-f12

the problem persists and now I have a brick.

I will try to boot a live cd with a new kernel to see if is something wrong with my ssd.

mercuriete, 2021-08-19:
After booting an ubuntu live cd and realising this computer it is indeed working...
pasing fsck to the main partition and checking all my files were correct...

The following steps worked for me:
Press f12 to open boot menu
try all options.
one of them is boot options recovery.

after that a new uefi entry appears called "ubuntu"

boot to the new entry and voilá the laptop is again working.

So in the end this BIOS update somehow broke my UEFI boot entries I it kept in a non-updated-state

mercuriete, 2021-08-19:
last update:

I updated again the firmware using fwupd from 1.8.0 to 1.9.0
but this time I used the console instead of the gui.

The update process was flawlessly.

I hope this helps somebody with my same problem.

hughsie, 2021-08-20:
It sounds like the firmware upgrade triggered some kind of cleanup. I'm not sure what exactly happened, but I'm glad it's working now!

mercuriete, 2021-08-20:
@hughsie you can close this issue if you want
Because there was no error on the Fwupd side and there was no error on the new bios itself.

The problem is on the Dell side or bios side when you don't have a bootable entry... It was booting to the entry called firmware update even when the update was completed and the UEFI capsule was removed (I guess).
In my opinion, it is good that Dell ships a tool called "boot entry recovery" or something like that (I don't recall). But it is not user-friendly that the laptop won't recreate the missing entries after an entry clean up.

Given said that, Thank you very much for this awesome tool. I think Fwupd is the best tool we have in the Linux world.

Thank you for your work.