fwupd/firmware-dell #85: Version comparison error: "Specified firmware is older than installed '0x15449bc < 22301115'"

leogama, 2021-08-31:
**Describe the bug**
This is similar to Similar to fwupd/fwupd#3217. `fwupdmgr` compares the firmware update version (22301116) to the installed version (22301115), concludes that the update is older than the current firmware and refuses to install it. The informative message (both in GNOME Software and in `fwupdmgr`) says that 22301115 would be upgraded to 22301116, showing both in decimal form. But the error message shows just the higher version number in hexadecimal form and the other in decimal.

GNOME Software capture:
[image]

Command-line outputs:
```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • DELL08C8:00 04F3:30C3
 • System Firmware
 • UEFI dbx
 • WDC WD10SPZX-21Z10T0
Vostro 3480
│
└─CL1-3D128-Q11 NVMe SSSTC 128GB:
  │   Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
  │   Summary:            NVM Express Solid State Drive
  │   Current version:    22301115
  │   Vendor:             Intel Corporation (NVME:0x1E95)
  │   GUIDs:              c37ff45c-9789-5d14-a8a6-cb1d664cca9c
  │                       1cd7457a-27a9-4af9-912b-73af953d94b4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─NVM Express Solid State Drive for CL1 Device Update:
        New version:      22301116
        Remote ID:        lvfs
        Summary:          Non SED NVME
        License:          Proprietary
        Size:             1,1 MB
        Created:          2021-01-14
        Urgency:          High
        Vendor:           SSSTC
        Flags:            is-upgrade
        Description:      
        Fixed the issue where the system does not reboot after enabling verifier mode.
        
        Improved the SSD performance and enhanced the battery life by reducing the power consumption.
```

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • DELL08C8:00 04F3:30C3
 • System Firmware
 • UEFI dbx
 • WDC WD10SPZX-21Z10T0
Upgrade available for CL1-3D128-Q11 NVMe SSSTC 128GB from 22301115 to 22301116
Vostro 3480 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Specified firmware is older than installed '0x15449bc < 22301115'
```

I suspect one of the numbers is in the wrong format internally, and/or is being silently coerced to an inappropriate type instead of raising an error.

**Steps to Reproduce**
Get a Dell notebook with an Intel NVMe SSD, wait for a firmware update for it, try to install the update either via GNOME Software or via `fwupdmgr update` command.

**Expected behavior**
`fwupd` could either recognize that `22301116` and `0x15449bc` are numbers and the same number, or identify the problem and show a meaningful error message. Note: I don't know how is the firmware version specification or the software implementation, so don't know if it's possible.

**fwupd version information**
Please provide the version of the daemon and client.
[... 114 lines cut ...]
│   
└─WDC WD10SPZX-21Z10T0:
      Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
      Summary:            ATA Drive
      Current version:    02.01A02
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      GUIDs:              fa8b4ca0-786c-54ea-9bd6-d025389b0590
                          807552eb-11ba-5e03-8894-84dad0e329a6
                          18d6d543-e351-5d90-a1c4-f0d3429463a2
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**

- Operating system and version: Ubuntu Linux 20.04 (up-to-date)
- Have you tried rebooting? No.
- Is this a regression? I think it was the first firmware update attempt in this machine:
```
$ fwupdmgr get-history                   
No history
```

leogama, 2021-08-31:
Indeed, there's no conversion of numeric format by `fwupd` as I tought. The code seems to treat versions mostly as strings internally. The version read from LVFS API (decimal) is the one used in the informative messages and, after user confirmation, the CAB file is downloaded and the version from it's metadata (hexadecimal) is used in the comparison and the error message.

Anyway, shouldn't `fwupd` be cross checking these metadata somewhere, either at upload to LVFS or just after download? Or is the info in the database generated _from_ the file?  🤔 

Thank you for your time.

superm1, 2021-08-31:
>Anyway, shouldn't fwupd be cross checking these metadata somewhere, either at upload to LVFS or just after download? Or is the info in the database generated from the file? 🤔

I don't disagree.  This seems like a pretty straightforward safety check before promoting to stable to check if the string has a 0x then it's probably not `VersionFormat` of number.
Would you mind to file a bug report for LVFS here to help prevent it in the future: https://gitlab.com/fwupd/lvfs-website

thejpster, 2021-08-31:
I fell over this exact issue with the exact same NVME drive in a relatively new Dell Latitude.

leogama, 2021-08-31:
> I don't disagree. This seems like a pretty straightforward safety check before promoting to stable to check if the string has a 0x then it's probably not `VersionFormat` of number.
> Would you mind to [...]

Done: https://gitlab.com/fwupd/lvfs-website/-/issues/823

hughsie, 2021-09-04:
Fixed in https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1070 -- if Mario agrees I'll regen the .cab file on the LVFS.

leogama, 2021-09-04:
Great! As soon as it is available I can retry the installation and give you feedback.

hughsie, 2021-09-04:
Okaym deployed, and regenerated as https://fwupd.org/downloads/4e659c2cd0d3e3df27c3ec098008ccdec62976260ea1291c4fdcb02cff9e34c3-22301116.cab -- that'll appear in the metadata in about 4 hours from now.

hughsie, 2021-09-05:
@leogama can you try a `fwupdmgr refresh --force` then `fwupdmgr update` please?

thejpster, 2021-09-05:
```
$ sudo fwupdmgr refresh
Firmware metadata last refresh: 3 hours ago. Use --force to refresh again.
$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 1 local device supported
$ sudo fwupdmgr update         
Devices with no available firmware updates: 
 • DLL09D9:01 04F3:3147
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for CL1-3D512-Q11 NVMe SSSTC 512GB from 22321115 to 22321116
Inspiron 7400 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Specified firmware is older than installed '0x15497dc < 22321115'
```

hughsie, 2021-09-05:
Ohh, that's a different firmware -- sorry -- I've just set that to regenerate as well. Give it 2 hours and then try again please.

leogama, 2021-09-06:
I'm feeling we are almost there:
```
Firmware version formats were different, device was 'plain' and release is 'number'
```

hughsie, 2021-09-06:
Can you try again now please? Thanks.

leogama, 2021-09-07:
I'm still getting the same file from 3 days ago (https://fwupd.org/downloads/4e659c2cd0d3e3df27c3ec098008ccdec62976260ea1291c4fdcb02cff9e34c3-22301116.cab), even after `refresh --force`.

thejpster, 2021-09-07:
I did an update, and it complained about the checksum being wrong. I did a refresh, then an update, and it worked fine. Rebooted my machine, and I still have an SSD at least!

hughsie, 2021-09-07:
@leogama I've fixed up com.dell.guid1cd7457a.firmware too -- wait 4h and then try again please.

leogama, 2021-09-08:
I've finally got it today's morning. It installed flawlessly and I uploaded the report.
```
Devices that have been updated successfully:

 • CL1-3D128-Q11 NVMe SSSTC 128GB (22301115 → 22301116)
```
Thank you, everybody, for the support. 😄 
Closing...

closed 2021-09-08 by leogama