fwupd/firmware-dell #88: Dell Inspiron 7577/Vostro 7570: outdated BIOS

sir-sigurd, 2021-09-28:
* latest BIOS is 1.13.0: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=8fm60&oscode=biosa&productcode=inspiron-15-7577-laptop
* latest BIOS at LVFS is 1.6.1: https://fwupd.org/lvfs/devices/com.dell.uefi7d0d2c9e.firmware
* system: Dell Inspiron 7577

sir-sigurd, 2021-11-16:
1.15.0 is available: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=gv69h&oscode=biosa&productcode=inspiron-15-7577-laptop

sir-sigurd, 2022-01-25:
1.16.0 is available: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=mcc0g&oscode=biosa&productcode=inspiron-15-7577-laptop&lwp=rt

pjwelsh, 2022-08-09:
Very sad to report that there has been a 1.17.0 version released:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=3mt9j&oscode=biosa&productcode=inspiron-15-7577-laptop
@dell-client-linux any chance to get this updated? It's been almost 1 year since original posting request.

CragW, 2022-08-10:
I thought 1.17.0 is avaialble at LVFS, see: https://fwupd.org/lvfs/devices/com.dell.uefi7d0d2c9e.firmware

pjwelsh, 2022-08-10:
@CragW thank you very much for noting that information... I should have checked first. Thank you.

@dell-client-linux -I wish the Dell BIOS updaters looked at the tickets to provide more feedback to when items are fixed or updated. I often find my "missing" requests are either silently updated based on my ticket or randomly updated... not sure which.

@sir-sigurd please confirm your OK for this ticket to be closed since the up-to-date BIOS is in place.

closed 2022-08-10 by sir-sigurd