fwupd/firmware-dell #93: Trying to update TPM firmware on 9360 gives me "Blocked by TPM FW Policy" error from firmware

mjg59, 2021-11-09:
Trying to either update the TPM 1.2 firmware to the latest, or trying to install TPM 2.0 firmware results in the system rebooting, starting the update, and then producing a "Blocked by TPM FW Policy" error. The same message is generated when trying to perform the update from DOS, so this isn't intrinsically an fwupd issue.

Dell XPS 13 9360.

hughsie, 2021-11-30:
@superm1 any ideas here? I know it's not your problem any more, but I figured you might know who to ping.

superm1, 2021-11-30:
I would suggest trying to clear the TPM in BIOS setup first and trying it again

mjg59, 2021-11-30:
@superm1 Tried that, no change.

superm1, 2021-11-30:
> @superm1 Tried that, no change.

And this update is intended for this system / tpm right?

If so, then Dell needs to repro and look further into it to solve.

therealjuanmartinez, 2021-11-30:
Agree - escalating this to internal TPM group

mjg59, 2021-11-30:
@therealjuanmartinez:

```sudo fwupdmgr get-releases c6a80ac3a22083423992a3cb15018989f37834d6
XPS 13 9360
│
└─TPM 1.2:
      New version:        5.81.2.1
      Remote ID:          lvfs
      Summary:            Firmware for the Dell TPM 1.2
      Licence:            Proprietary
      Size:               554.9 kB
      Created:            2016-12-02
      Urgency:            Low
      Vendor:             Dell
      Flags:              is-upgrade
      Description:        
      Initial release

$ sudo fwupdmgr upgrade c6a80ac3a22083423992a3cb15018989f37834d6
TPM 1.2 and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating TPM 1.2…        [***************************************]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]:```

After reboot, gives me the "Update failed: Blocked by TPM Policy" error. 

``` sudo fwupdmgr install https://fwupd.org/downloads/a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab 
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Installing on TPM 2.0…   [***************************************]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]:```

fails with the same error. get-history gives:

```└─TPM 2.0:
  │   Device ID:          a60b665c769b019e30f31001e91ab31e24d7a035
  │   Previous version:   0.0.0.0
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
```

therealjuanmartinez, 2021-11-30:
@mjg59 Could you attach the output (or at least the TPM portion(s)) of the following?: 

`fwupdmgr get-devices --filter=internal,~supported`

superm1, 2021-11-30:
(Presumably you want the current BIOS version too - in case there is something to do with the BIOS version on the machine controlling this policy)

mjg59, 2021-12-01:
```
XPS 13 9360
│
├─CT500MX500SSD4:
│     Device ID:          64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:            ATA Drive
│     Current version:    M3CR020
│     Vendor:             Micron (ATA:0x1344, OUI:00a075)
│     GUIDs:              cd4b908b-edef-5a08-8616-463a2c739755
│                         a67a9709-2b42-519a-9760-5ede74ce8609
│                         e9b83e95-968d-5310-aa2d-ab745ca18d87
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          3f0ecc7823ee99c718693e8ecca38f54e8738dc5
│ │   Current version:    2.18.0
│ │   Minimum Version:    2.18.0
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              5ffdbc0d-f340-441c-a803-8439c8c0ae10
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       66d5d415-835d-561c-b687-dcfe4c8b2558
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
└─TPM 1.2:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            Platform TPM device
      Current version:    5.81.0.0
      Vendor:             Dell Inc. (PCI:0x1028)
      Update State:       Success
      GUIDs:              f9bdd338-b410-5e73-902d-7b6e4694bb56
                          ff71992e-52f7-5eea-94ef-883e56e034c6
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
```

therealjuanmartinez, 2021-12-02:
The DXE driver is the likely culprit for disallowing the update between those particular TPM FW versions, which is typical when seeing the "Blocked by TPM FW Policy" error.

The FW version you need is 1.3.2.8, and DXE should allow a direct upgrade from the current version.  Additionally, it seems you're on a recent/latest BIOS for your platform so that should not be an issue.

Unfortunately, there are no capsule builds available (via LVFS or otherwise) due to the fact that it was published at a time before Dell was fully qualifying most/all TPM releases for Linux release.

Therefore, my suggestion is to retrieve the TPM FW v1.3.2.8 directly from Dell.com for the XPS 13 9360, and use a WinPE key to facilitate installation.

mjg59, 2021-12-02:
I tried running the 1.3.2.8 update from https://www.dell.com/support/home/en-us/drivers/DriversDetails?driverId=0DJC8 under FreeDOS (the 32-bit version does not require Windows) and it generates the same error.

therealjuanmartinez, 2021-12-07:
Nuvoton has verified the upgrade path from 5.81.0.0 (TPM1.2) -> 1.3.2.8 (TPM2.0) is allowed from FW perspective.

As a result, it's unclear whether they are mistaken or there is something unique happening on your side.

The FW utility link you shared in his issue is an older utility, the latest for FW v1.3.2.8 for his platforms is: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=8n08g&oscode=wt64a&productcode=xps-13-9360-laptop

You MAY need to run WinPE for this because FreeDOS may not be supported by this FW Update wrapper anymore.

Alternatively there is another TPM1.2 FW Update (though not officially published for your platform): https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=twhk9&oscode=wt64a

It should be supported based on the FW you have now.  This is TPM1.2 FW v5.81.2.1, which is a newer TPM1.2 FW to the v5.81.0.0 on your system. Could you give this one a try? Once again, might need to run in WinPE directly.

If the above still doesn't help, let me know and we can look further.

mjg59, 2021-12-09:
Same failure with both the 1.3.2.8 and the 5.81.2.1 updates.

therealjuanmartinez, 2021-12-09:
Can you confirm that the TPM is not owned?

mjg59, 2021-12-09:
I'm clearing the TPM in the firmware before each attempt

therealjuanmartinez, 2021-12-15:
Just for clarity can you again confirm the error is still the "Blocked by TPM FW Policy" message, and that you're using WinPE?

mjg59, 2021-12-16:
Yes to both

therealjuanmartinez, 2022-01-27:
Hi again, I wanted to let you know we're currently working with the component vendor to find whether there are commands that can be run on your end to pull some specific diagnostic information we're looking for.  Where we go from there will depend on their response.

therealjuanmartinez, 2022-02-28:
Hi again - this is probably a long shot but is there a chance you are still using the same factory install on that machine?  And if not, by chance might you have made OEM Ubuntu recovery media using the software tools shipped with that device?  The reason I ask is because recovery partition files would make it easier for us to learn what exactly took place during the factory process for that machine.  We're still pursuing the action from my last message.

mjg59, 2022-03-01:
I'm afraid not - I reimaged it with Fedora. I'm not certain, but I believe I may have switched it between TPM 1.2 and 2.0 a couple of times for development testing in the past.

fredoche, 2022-09-12:
FWIW I apparently have the same issue with a tpm 1.2 blocked at 5.81.0.0.
Sometimes the GUI "Firmware" reports a version of 255.255.255.255 after clearing or unlocking the TPM.
[image]

hugh712, 2022-09-16:
@fredoche 

Would you please provide the dmidecode log once you observe this issue?
`sudo dmidecode > dmidecode.log`

fredoche, 2022-09-17:
`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
94 structures occupying 5806 bytes.
Table at 0x000E0000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Dell Inc.
	Version: 1.13.0
	Release Date: 02/10/2020
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.13

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Dell Inc.
	Product Name: XPS 13 9350
	Version: Not Specified
	Serial Number: 4HB5Q72
	UUID: 4c4c4544-0048-4210-8035-b4c04f513732
	Wake-up Type: Power Switch
	SKU Number: 0704
	Family: NULL

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Dell Inc.
	Product Name: 0H67KH
	Version: A05
	Serial Number: /4HB5Q72/CN1296367K0177/
	Asset Tag: Not Specified
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Dell Inc.
	Type: Laptop
	Lock: Not Present
	Version: Not Specified
[... 980 lines cut ...]
Handle 0xF03C, DMI type 219, 81 bytes
OEM-specific Type
	Header and Data:
		DB 51 3C F0 01 03 01 55 02 00 A0 06 01 00 66 20
		00 00 00 04 40 08 00 01 1F 00 00 C9 0B C0 47 02
		FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
		FF FF FF FF FF FF FF FF 03 00 00 00 80 00 00 00
		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
		00
	Strings:
		MEI1
		MEI2
		MEI3

Handle 0xF03D, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 2
		en|US|iso8859-1
		<BAD INDEX>
	Currently Installed Language: en|US|iso8859-1

Handle 0xF03E, DMI type 127, 4 bytes
End Of Table
`

fredoche, 2022-09-17:
Dmidecode when UI shows 5.81.0.0:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
94 structures occupying 5806 bytes.
Table at 0x000E0000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Dell Inc.
	Version: 1.13.0
	Release Date: 02/10/2020
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.13

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Dell Inc.
	Product Name: XPS 13 9350
	Version: Not Specified
	Serial Number: 4HB5Q72
	UUID: 4c4c4544-0048-4210-8035-b4c04f513732
	Wake-up Type: Power Switch
	SKU Number: 0704
	Family: NULL

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Dell Inc.
	Product Name: 0H67KH
	Version: A05
	Serial Number: /4HB5Q72/CN1296367K0177/
	Asset Tag: Not Specified
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Dell Inc.
	Type: Laptop
[... 984 lines cut ...]
	Header and Data:
		DB 51 3C F0 01 03 01 55 02 00 A0 06 01 00 66 20
		00 00 00 04 40 08 00 01 1F 00 00 C9 0B C0 47 02
		FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
		FF FF FF FF FF FF FF FF 03 00 00 00 80 00 00 00
		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
		00
	Strings:
		MEI1
		MEI2
		MEI3

Handle 0xF03D, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 2
		en|US|iso8859-1
		<BAD INDEX>
	Currently Installed Language: en|US|iso8859-1

Handle 0xF03E, DMI type 127, 4 bytes
End Of Table

```
I'll try to reproduce the issue with the weird version and post the result here.

fredoche, 2022-09-17:
To reproduce, I clicked on "upgrade" in the firmware UI to update to 5.81.2.1 . The UI asked me to reboot, I clicked "Later" and now the UI shows version 255.255.255.255. 
dmidecode:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.
94 structures occupying 5806 bytes.
Table at 0x000E0000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Dell Inc.
	Version: 1.13.0
	Release Date: 02/10/2020
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.13

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Dell Inc.
	Product Name: XPS 13 9350
	Version: Not Specified
	Serial Number: 4HB5Q72
	UUID: 4c4c4544-0048-4210-8035-b4c04f513732
	Wake-up Type: Power Switch
	SKU Number: 0704
	Family: NULL

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Dell Inc.
	Product Name: 0H67KH
	Version: A05
	Serial Number: /4HB5Q72/CN1296367K0177/
	Asset Tag: Not Specified
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Dell Inc.
[... 984 lines cut ...]
OEM-specific Type
	Header and Data:
		DB 51 3C F0 01 03 01 55 02 00 A0 06 01 00 66 20
		00 00 00 04 40 08 00 01 1F 00 00 C9 0B C0 47 02
		FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
		FF FF FF FF FF FF FF FF 03 00 00 00 80 00 00 00
		00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
		00
	Strings:
		MEI1
		MEI2
		MEI3

Handle 0xF03D, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 2
		en|US|iso8859-1
		<BAD INDEX>
	Currently Installed Language: en|US|iso8859-1

Handle 0xF03E, DMI type 127, 4 bytes
End Of Table

```

fredoche, 2022-09-17:
After reboot and a tpm upgrade failure "blocked by tpm fw policy", the UI still shows version 255.255.255.255 
and even offers me to "downgrade" to 5.81.2.1.

hugh712, 2022-09-17:
hmmmm, there is no tpm info in dmidecode...
let me see if I can find a XPS 13 9360 to reproduce this issue

fredoche, 2022-09-17:
btw this is on a 9350.

fredoche, 2022-09-17:
In the TPM 2 section there is a "Lock status" section. When I click to unlock, the GUI tells me to reboot, yet, after reboot, the GUI has the "Lock status" locked again like nothing happened.

tim-seoss, 2022-09-20:
I'm seeing the same issue on an Optiplex 5040.  Current TPM 1.2 version 5.81.0.0 attempting to update to 5.81.2.1.

I then tried installing the TPM 2.0 update from Window 10, but this also failed in the same way.

hugh712, 2022-09-27:
@tim-seoss 

Please help to contact dell support on the website [0] ,
I believe it's a bios issue not fwupd issue since even window is failed to do it, thank you.

[0] [https://www.dell.com/support](https://www.dell.com/support)

tim-seoss, 2022-09-27:
@hugh712 Thanks for the suggestion, but given that this problem appears to exist across multiple Dell models with the same firmware update...  My assumption is that there is an interaction between the firmware update and the platform firmware that requires fixing.

At the very least it needs to have a more helpful error message than "Blocked by TPM FW Policy". My assumption is that you would have orders of magnitude more chance of getting this problem fixed from within Dell, by referring it to the platform firmware team than I ever would as a individual customer in possession of a single machine.

fredoche, 2022-09-27:
Though I have contacted Dell support but my laptop being out of warranty, support cannot help. @therealjuanmartinez mentionned being in contact with vendor, about this very issue, which seems promising, but maybe the chip support is simply abandonned.

fredoche, 2022-10-22:
Hmm, and today, after a round of updates, fwupdmgr does not list the TPM anymore...
```
[root@cinderblock fred]# fwupdmgr get-devices
Dell Inc. XPS 13 9350
│
├─Core™ i7-6560U CPU @ 2.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000f0
│     Fournisseur:        Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         28d7f31c-1663-5b15-a900-94a71cbcd53f ← CPUID\PRO_0&FAM_06&MOD_4E
│                         e4bc7423-9fec-55a6-8f6c-827c40992491 ← CPUID\PRO_0&FAM_06&MOD_4E&STP_3
│     Drapeaux de périphérique:• Périphérique interne
│   
├─Iris Graphics 540:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Version actuelle:   0a
│     Fournisseur:        Intel Corporation (PCI:0x8086)
│     GUIDs:              7b246578-7083-57a2-aad3-f421e8ba9ed0 ← PCI\VEN_8086&DEV_1926
│                         1aa9bc07-3911-5ba9-9ede-7efb78dda7e9 ← PCI\VEN_8086&DEV_1926&REV_0A
│                         37dbff15-684f-5799-ae0d-591edbb218fe ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704
│                         18b3de8b-181c-5dcd-a69d-484d79511211 ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704&REV_0A
│                         c568e060-8b86-5a8c-a86e-56060b784495 ← PCI\VEN_8086&DEV_1926&REV_00
│                         26e22885-57dc-520e-b290-2e727ebff8be ← PCI\VEN_8086&DEV_1926&SUBSYS_10280704&REV_00
│     Drapeaux de périphérique:• Périphérique interne
│                         • Cryptographic hash verification is available
│   
├─PM951 NVMe SAMSUNG 512GB:
│     Device ID:          e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Résumé:             NVM Express solid state drive
│     Version actuelle:   BXV77D0Q
│     Fournisseur:        Samsung (NVME:0x144D)
│     Numéro de série:    S29PNXAH214218
│     GUIDs:              77bb4022-9503-586c-99b7-c8a64502da30 ← NVME\VEN_144D&DEV_A802
│                         cce2386f-6d68-5f85-9f3f-ebead9288696 ← NVME\VEN_144D&DEV_A802&REV_01
│                         484eb66f-4175-5499-8dc9-a8cc651e8450 ← NVME\VEN_144D&DEV_A802&SUBSYS_144DA801
│                         b586caf2-ac7e-5a07-8630-dd289d8fd3f4 ← NVME\VEN_144D&DEV_A802&SUBSYS_144DA801&REV_01
│                         2b05215d-0ecf-5749-a584-012329fb33e6 ← PM951 NVMe SAMSUNG 512GB
│     Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│                         • Le système nécessite une source d'alimentation externe
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Résumé:             UEFI ESRT device
  │   Version actuelle:   1.13.0
  │   Version minimum:    1.13.0
  │   Fournisseur:        Dell (DMI:Dell Inc.)
  │   État de mise à jour:Success
  │   GUIDs:              33773727-8ee7-4d81-9fa0-57e8d889e1fa
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Drapeaux de périphérique:• Périphérique interne
  │                       • Mise à jour possible
  │                       • Le système nécessite une source d'alimentation externe
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Résumé:           UEFI revocation database
        Version actuelle: 217
        Version minimum:  217
        Fournisseur:      UEFI:Linux Foundation
        Durée d'installation:1 seconde
        GUIDs:            c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Drapeaux de périphérique:• Périphérique interne
                          • Mise à jour possible
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
      
En attente…              [***************************************]

```
though the tpm appears in /dev/tpm0:
```
[root@cinderblock fred]# ls /dev/tpm0 
/dev/tpm0
```

fredoche, 2023-02-19:
Just tried again today, and while it still doesnt work, I noticed this in the fwupdmgr after a failed tentative:
```
└─TPM 1.2:
  │   Device ID:          a3487e128cf1413519bce8e9a1ab3f5981e61458
  │   Résumé:             UEFI ESRT device
  │   Version actuelle:   5.81.0.0
  │   Fournisseur:        Dell Inc. (PCI:0x1028)
  │   État de mise à jour:Success
  **│   Erreur de mise à jour:Preventing upgrades as alternate**
  │   GUIDs:              d433959e-03ca-524b-92b7-5022eff81a31 ← 0704-1.2
  │                       ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
  │   Drapeaux de périphérique:• Périphérique interne
  │                       • Le système nécessite une source d'alimentation externe
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Mise à jour possible
  │ 
  └─TPM 2.0:
        Device ID:        eb4a4ec71c680a34c6248dc94ca017f81c772aa8
        Résumé:           UEFI ESRT device
        Version actuelle: 0.0.0.0
        État de mise à jour:Success
        Dernière modification:2023-02-19 14:33
        GUID:             b62a2412-5ac4-5350-b16e-7e8f4655d096 ← 0704-2.0
        Drapeaux de périphérique:• Périphérique interne
                          • Mise à jour possible
                          • Le système nécessite une source d'alimentation externe
                          • Supported on remote server
                          • Needs a reboot after installatio
```
Notive "Preventing upgrades as alternate". What does it mean?

@hughsie going out on a limb here, yet  I'm wondering if the "policy" that blocks update is "Since there is a tpm 2 update, dont bother update the TPM 1.2 and instead you should update to tpm 2 instead" ? And since fwupmgr keeps trying to push the tpm 1.2 update instead of the 2.0, it will always fail?  So, could it be that the problem is an UI one, and that, somehow, if fwupdmgr could offer the update to 2.0 (skipping the whole 1.2 thing altogether), the upgrade would work ?

xyzkernel, 2023-11-22:
Today I encountered the same problem on my [Precision 7920 Tower]. Tried switching between versions, with no success. I contacted customer service and they said they couldn't update. But I think DELL should have set up a protection program in the module and refused to update.It is the aforementioned [**Preventing upgrades as alternate**].