fwupd/firmware-dell #94: Cannot even enter BIOS after update

WhymustIhaveaname, 2021-11-14:
**Describe the bug and Steps to Reproduce**
Today when I boot my Ubuntu 20, it tells me that there is a new firmware update available. And I can check by some command like (I cannot remember it exactly) `fw.... get-updates`. My computer is a Dell Optiplex 7070. I check the update, it says some optimize in fan control. I thought it was good, then I typed `fw... update` and the update went sommthly and finished successfully. Then reboot and BOOM! No Logo, no bios, nothing!

The light of my computer is 2 orange and 5 white, after consulting the datasheet, is "bios checksum error". I tried reboot, I tried to get rid of bios battery for a while and then reboot. I tried to rewrite bios using a USB stick but also failed. What can one do if one cannot access bios?!

**fwupd version information**

the computer cannot boot now

**Additional questions**

- Operating system and version: Ubuntu 20
- Have you tried rebooting? Yes, definitely
- Is this a regression? what?

hughsie, 2021-11-14:
@dell-client-linux hey can you look at this as a prio please.

dell-client-linux, 2021-11-15:
Which BIOS version? The latest one in LVFS?

WhymustIhaveaname, 2021-11-15:
Thank you for your prompt reply. I cannot access BIOS so far, so I do not know which version the bios is. However, as the system alerts me to update, I think it is not the latest one. I vaguely remembered that there are two points in the update message, now of them is optimizing fan behavior for better cooling.

I tried to contact Dell for help. It asks me to provide the service number. But the service number for this machine is in the bios! I cannot open bios, so no number, so cannot bypass that stupid AI to contact a true engineer.

dell-client-linux, 2021-11-15:
@WhymustIhaveaname, you can find the service number (Service Tag) from the physical label attached on the machine. 

Checked the latest BIOS in LVFS for Optiplex 7070 and it's 1.10.0. Did not get any failure reports from others. Therefore, should still need to check the flash log from your system after the system recovered. 

Please contact Dell Service team to recover your system again after you find the Service Tag from the label.

WhymustIhaveaname, 2021-11-15:
Yes, normally there is a service number somewhere on the machine. But for this one, optiplex 7070 MFF, this is no longer true. I owned 4 or 5 Dell devices and I am sure that I looked up every corner. I also asked the guy who sold me this machine (Yes, this is a second hand machine) and he confirmed this. The only way of getting service number is from bios.

I called Dell and ask for service, but they refuses to provide service because I cannot give them the service number (can I sue Dell for this?). This is a dead-lock now.

hughsie, 2021-11-15:
> I also asked the guy who sold me this machine 

Do they have a record of the service number? Or any kind of serial number you could use? It does sound like hardware failure given what @dell-client-linux says.

WhymustIhaveaname, 2021-11-15:
Neither of us recorded the sevice number. Because we thought dell's product noramlly do not fail.

This machine has been working for a month until that day I update.

I think no one report this peoblem because the users of optiplex is fewer. There are far less discussions about this mathice than other PCs. This is also the painful part: I cannot find enough information to help me reinstall the bios. There are some jumpwires on the motherboard which may work but I cannot find out their meaning. It has been two days since the disaster. I still cannot fix it.