fwupd/firmware-dell #95: Dell XPS 13 9365 failed to run update on reboot

andremarcais, 2021-11-22:
**Describe the bug**
Greetings,

The update for System Firmware is not applied after rebooting. When the system is rebooted to finish the update, it gives the error: "Hard drive not installed" (or something in the likes of that).

This has been going on for a while, and my recent attempt to fix it (detailed below) was unsuccessful.

**Steps to Reproduce**
I try to update the system following the Arch Wiki's [instructions](https://wiki.archlinux.org/title/Fwupd). When I am prompted to reboot, I reboot the system. It fails to boot the firmware updater, and afterwards when I boot Linux, the firmware is not updated.
```
$ fwupdmgr get-history
XPS 13 9365
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   2.17.0
  │   Update State:       Failed
  │   Update Error:       [31m[1mfailed to run update on reboot[0m
  │   Last modified:      2021-11-19 19:50
  │   GUID:               d636245a-c31b-4454-b9a5-9893c6e6f6cc
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─XPS 13 9365:
        New version:      2.18.0
        Remote ID:        lvfs
        Summary:          Firmware for the Dell XPS 13 9365
        License:          Proprietary
        Size:             14.8 MB
        Created:          2021-10-08
        Urgency:          Critical
        Vendor:           Dell
        Flags:            is-upgrade
        Description:      
        Some of the platform secrets may be invalidated when updating this firmware. Please ensure you have the volume recovery key before continuing.
        
        This stable release fixes the following issues:
        
        • Firmware updates to address the Intel Security Advisory.
      

```

Before I was updating from something like 1.3.x, so I figured that the firmware on my system was to old to be supported by fwupd. So I manually updated to version 2.17.0 from the F12 menu using Dell's [package](https://www.dell.com/support/home/fr-fr/drivers/driversdetails?driverid=48w05&oscode=biosa&productcode=xps-13-9365-2-in-1-laptop), purposefully updating to the version before the latest (2.18.0) hoping to get fwupd perform an update. The manual update works, but fwupd still fails to update after following the Arch Wiki instructions again. (Now it doesn't really give me an error. Instead the BOIS diagnostic tool starts up and prompts me to download cSOS, to which I decline. I presume it's still the same problem as before the manual update.)

**Expected behavior**
The firmware updater is able to boot without error, and when It's done my system firmware is up to date.

**fwupd version information**
```
$ fwupdmgr --version
client version:	1.7.1
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.1
```

I installed fwupd with pacman on Arch Linux.

**fwupd device information**
Here are the fwupd devices recognized in my system:

```
$ fwupdmgr get-devices --show-all-devices
[... 129 lines cut ...]
    │   Device Flags:     • Internal device
    │                     • System requires external power source
    │                     • Device is locked
    │ 
    └─TPM 1.2:
          New version:    89195009
          Remote ID:      lvfs
          Summary:        Firmware for the Dell TPM 1.2
          License:        Proprietary
          Size:           554.9 kB
          Created:        2016-12-02
          Urgency:        Low
          Vendor:         Dell
          Description:    
          Initial release
        

```

**Additional questions**

- Operating system and version:
   `Linux cpt-pex-dell-laptop 5.15.2-zen1-1-zen fwupd/fwupd#1 ZEN SMP PREEMPT Fri, 12 Nov 2021 19:22:07 +0000 x86_64 GNU/Linux`
- Have you tried rebooting? Yes, as required for the update.
- Is this a regression? No.

andremarcais, 2021-11-23:
Sure. When I boot fwupdx64.efi from the shell it says it can't "query capsule capabilities." Attached is a video of that. (Warning: the video includes my computer's very loud beep.)

When I select the "Linux Firmware Updater" boot entry from the GUI it immediatly beeps and prompts me to retry booting. (Unfortunately, I do not have a video of the "Hard Drive not Installed" error it gave me before manually updating to the more recent firmware.)

https://user-images.githubusercontent.com/47303971/143061289-fb455679-2751-457c-874b-ab7df8077b09.mp4

hughsie, 2021-11-23:
@dell-client-linux can you escalate the `query capsule capabilities` error please? This is where RT->QueryCapsuleCapabilities returns UNSUPPORTED in https://github.com/fwupd/fwupd-efi/blob/main/efi/fwupdate.c#L455

andremarcais, 2021-11-28:
Hey, any update on this?

I still haven't updated my firmware. Do you have the info you need to fix the problem?

Thanks.

dell-client-linux, 2021-11-29:
@andremarcais, to temporary workaround this issue and update BIOS in your system, please help download the BIOS from Dell.com and flash the BIOS through BIOS menu (press F12 during reboot).

pjwelsh, 2022-02-20:
@andremarcais Not sure if it helps, but historically when I have a BIOS that does not like to apply, "normally", somehow with just the additional command run as user root after the "fwupdmgr update" of:
`printf "\x07\x00\x00\x00\x01" > /sys/firmware/efi/efivars/FWUPDATE_VERBOSE-0abba7dc-e516-4167-bbf5-4d9d1c739416`
the update will magically proceed in many cases on it's own. This has no chance of helping if you get a "file not found" kind of message, however. It happens(ed?) so much that it's just part of my update scripting.
I don't recall where I picked that command up from, but likely from here somewhere.

hughsie, 2022-02-21:
> printf "\x07\x00\x00\x00\x01"

Hmm, that's the same effect of `sudo fwupdate --set-debug` -- right? I'm confused why that would make any difference; the only change it might be making is that everything takes a few more ms to process because it's saving the debug log to flash.

pjwelsh, 2022-02-21:
Could be the same as `--set-debug`. I just know about using it based on some past issues/google'n.
But in *many* cases were `fwupdate update` will place the files like it should work an the BIOS update process will begin and *immediately* fail out (with no readable message), *just* doing the `printf` (or --set-debug?) will *succeed* (and produce debug output). I don't know why the `printf` worked. Ultimately I just cared about getting the BIOS updated more than trying to capture the debug code of a BIOS update that ended up *actually* working (with the debug on).

pjwelsh, 2022-02-21:
I should note that there are cases where the `fwupdate update` appears to work, but the `/sys/firmware/efi/efivars/FWUPDATE_VERBOSE-0abba7dc-e516-4167-bbf5-4d9d1c739416` location will not exist. IIRC, when that happens, I know a manual BIOS update will need to happen along with verifying TPM state (likely needing to clear). The scripting has a pre-check to validate that the "firmware capsule enabled" BIOS option is checked for.

andremarcais, 2023-01-28:
I'm closing this issue, as it has been resolved a while ago.  There was a problem with my efivars.  Not sure what caused the problem, but manually setting the boot option for fwupd solved it. (IIRC the partition UUID wasn't right.)

closed 2023-01-28 by andremarcais