fwupd/firmware-dell #100: precision t3620 BIOS 2.19.0 missing

pjwelsh, 2021-11-29:
precision+t3620
LVFS version  2.18.1
https://fwupd.org/lvfs/search?value=precision+3620
Dell version 09 Nov 2021,2.19.0, 2.19.0
https://www.dell.com/support/home/en-us/product-support/product/precision-t3620-workstation/drivers

pjwelsh, 2022-01-31:
Now the next Urgent version has changed:
Dell version 07 Jan 2022,2.20.0, 2.20.0
https://www.dell.com/support/home/en-us/product-support/product/precision-t3620-workstation/drivers

pjwelsh, 2022-02-20:
@dell-client-linux please update to latest dell "Urgent".

pjwelsh, 2022-03-10:
Update still missing.

pjwelsh, 2022-04-12:
@dell-client-linux Please update as soon as possible. This will be the third update missed BIOS from the original post if not:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=yfjyj&oscode=wb64a&productcode=precision-t3620-workstation

sunnycrown83, 2022-04-12:
@pjwelsh under working, will solve it asap.

pjwelsh, 2022-04-26:
@sunnycrown83 any update on this missing firmware? Two weeks have passed since your response of "... will solve asap".

fuqian2022, 2022-04-28:
@pjwelsh , Sorry for the late response. 
BIOS team is working on uploading the new release to LVFS. Target release date: May 4th.

pjwelsh, 2022-04-28:
Thank you for the update.
You may already know, but a significant number of "urgent" BIOS updates were just released in the last day (optiplex-5060 optiplex-3040m optiplex-7060 optiplex-3060 optiplex-7070 optiplex-5070 etc)

pjwelsh, 2022-05-16:
@sunnycrown83 any update on the new timeline as the May 4th noted target has passed?

fuqian2022, 2022-05-19:
@pjwelsh, Sorry for the delay. 2.21.0 has been released to LVFS site.

pjwelsh, 2022-05-19:
I can confirm that the BIOS update both is showing and updates as expected on the systems I've tested with. Thank you very much for your help on this.

closed 2022-05-19 by pjwelsh