fwupd/firmware-dell #97: optiplex 5040 BIOS 1.20.0 missing

pjwelsh, 2021-11-29:
optiplex+5040
LVFS version  1.19.0
https://fwupd.org/lvfs/search?value=optiplex+5040
Dell version 09 Nov 2021,1.20.0, 1.20.0
https://www.dell.com/support/home/en-us/product-support/product/optiplex-5040-desktop/drivers

pjwelsh, 2022-01-31:
Now missing 1.21.0:
Dell version 12 Jan 2022,1.21.0, 1.21.0
https://www.dell.com/support/home/en-us/product-support/product/optiplex-5040-desktop/drivers

pjwelsh, 2022-02-20:
@dell-client-linux please update to latest dell "Urgent".

pjwelsh, 2022-03-10:
Update still missing.

pjwelsh, 2022-04-05:
<frustration=on>Common Dell, what is going on? (Dell people)I wish I could understand your update pattern... It seems that some systems will get immediate updates (Latitude 5590 for example) and other just languish forgotten/ignored like the half-dozen tickets since November I have open. To make matters worse, multiple "Urgent" updates get missed.
Please, someone tell me the magic foo to get some Dell priority for something that to the untrained end-user seems should very simple.
Heck even I came up with simple scripting to babysit a select list of Dell system BIOS updates -vs- fwupd updates to let me know when updates happen (or are missed). (reality note: basis for parts of scripting are from others with similar needs/issues)

dell-client-linux, 2022-04-07:
@pjwelsh, we are working with internal teams to solve the missing items. As the systems belong to different line of business and have different ODM, it would take time on some old platforms.

pjwelsh, 2022-04-07:
The time for "Urgent" firmware updates should not be 5 months (or more) and
miss more than one "Urgent" update.
Not sure what the process to add "official" firmware to fwupd is, but how
can it take so long? Isn't the firmware/BIOS payload
fundamentally already on a dell site?
As for ODM (Original Design Manufacturer?), I'm not even sure how that
plays into this issue. From the perspective the person buying the
equipment, these are Dell Latitude and Dell Optiplex systems that are
*still* supported with BIOS updates.
As far as "old platforms", I have at least one reported issue that is over
1 year old - a Latitude 3490/3590 that *was* new and in the middle of a
product platform - that somehow still has no BIOS.

We, as a business, have chosen to use Linux-to-the-desktop for the 250+
workstations and laptops (and servers) we have. I continue to purchase Dell
systems because of the value added by Dell's participation in projects like
this.
So, I very much appreciate what is being done. However, I believe that more
can be done. This would be especially true, by looking at the open tickets
for missing firmware and acting sooner, knowing there are people/companies
behind those tickets with issues.
Do these lingering issue make Dell look better or worse over time?

Regards
PJ

Disclaimer

The information contained in this communication from the sender is confidential. It is intended solely for use by the recipient and others authorized to receive it. If you are not the recipient, you are hereby notified that any disclosure, copying, distribution or taking action in relation of the contents of this information is strictly prohibited and may be unlawful.

This email has been scanned for viruses and malware, and may have been automatically archived.  Cassens

pjwelsh, 2022-04-11:
@dell-client-linux - looks like this has been (silently?) corrected. Thank you. closing.

closed 2022-04-11 by pjwelsh