fwupd/firmware-dell #103: Firmware for Dell Latitude 7320 Detachable is older then on the Dell homepage

ckruetze, 2022-01-05:
**Describe the bug**
fwupdmgr does not find a new firmware for a Dell Latitude 7320 Detachable.

**Steps to Reproduce**
Steps to reproduce the behavior.
Go to dell.com ( https://www.dell.com/support/home/de-de/product-support/servicetag/0-djFZTU5OQTZpN1ZlRlpqbkZGVXkxUT090/drivers ) and locate the latest firmware. as of writing this it is version 1.8.1 from 9th of December 2021.

**Expected behavior**
Find and update the firmware to the latest one available on the Dell homepage. I am not entirely sure but I think this https://fwupd.org/lvfs/devices/com.dell.uefi85686610.firmware is the one available for fwupd, so it would be older.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Latitude 7320 Detachable
│
├─BC711 NVMe SK hynix 512GB:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express Solid State Drive
│     Current version:    41002131
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      CD0CN91901140CT2E
│     GUIDs:              179a39a0-42d6-5c48-a12b-e3d681643334 ← STORAGE-DELL-110001
│                         0a413a0e-88f2-11ea-bc55-0242ac130003
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│   
├─Fingerprint Sensor:
│     Device ID:          84d0e3f2a0f8b2328f7995767b23ebb40494723f
│     Summary:            Match-On-Chip Fingerprint Sensor
│     Current version:    01000254
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UID3DA76133_XXXX_MOC_B0
│     GUIDs:              4395e6dc-636f-5cd9-bf48-20bb0d4fce6e ← USB\VID_27C6&PID_6384&REV_0100
│                         13982b83-7cb2-5f98-a607-30a0ab79a1ea ← USB\VID_27C6&PID_6384
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│   
├─System Firmware:
│ │   Device ID:          a85268a15f7a868e7778e4d3eb841a48602fe12a
│ │   Current version:    67328
│ │   Minimum Version:    67328
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              cf6c9f05-c7d8-4a0d-bbe2-10b48cebf582
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       a11197b9-bbed-5eb2-8bda-0c07f3640797 ← UEFI\RES_{CF6C9F05-C7D8-4A0D-BBE2-10B48CEBF582}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─UEFI Device Firmware:
│     Device ID:          6b9d757c1f64d696e7af69942c2d6ed3993479d4
│     Current version:    134
│     Minimum Version:    134
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              176e090d-0ddb-495e-8173-bc998ccfecd0
│                         3e1d83c5-31af-55ac-9f88-8be02bda9596 ← UEFI\RES_{176E090D-0DDB-495E-8173-BC998CCFECD0}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          97a677f8af7b0d44e649b577d1ab27f9fdfb5074
│     Current version:    1090527537
│     Minimum Version:    1090527537
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              0a413a0e-88f2-11ea-bc55-0242ac130003
│                         4ea0b1e1-6ee5-5875-8794-3754dc948ece ← UEFI\RES_{0A413A0E-88F2-11EA-BC55-0242AC130003}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          2376ddc27b6025c72cd519489fc5ce210eb77dec
      Summary:            A miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable

```

**Additional questions**

- Operating system and version: Ubuntu 20.04.3 LTS
- Have you tried rebooting?: Yes
- Is this a regression?: No ?

ckruetze, 2022-01-18:
Short update: dell.com now has version 1.9.0

ckruetze, 2022-03-24:
Another short update: dell.com is now at version 1.10.0

CragW, 2022-05-09:
This platform doesn't support Linux/Ubuntu therefore no firmware updates provided in fwupd/LVFS.
https://www.dell.com/support/home/en-us/product-support/product/latitude-13-7320-detachable-laptop/drivers

See if this script helps if you need an update badly.
https://github.com/fwupd/fwupd/blob/main/contrib/firmware_packager/install_dell_bios_exe.py