fwupd/firmware-dell #105: install_dell_bios_exe.py "GObject.connect requires at least 2 arguments" CentOS Stream (Fedora 34+)

pjwelsh, 2022-01-30:
Instructions for urgent BIOS updates from page https://github.com/fwupd/firmware-dell asks to run program "install_dell_bios_exe.py". This currently fails on CentOS 8 Stream (and Fedora 34+) with error:
Traceback (most recent call last):
  File "./install_dell_bios_exe.py", line 103, in <module>
    CLIENT.connect()

(Note this error is from latest git grab. My original workingcopy had the error on line 101) 

This appears to be related to "unintentional fwupd_client_connect API change between 1.6.1 and 1.6.2? #3644" from https://github.com/fwupd/fwupd/issues/3644
The indication is that the CLIENT.connect is extraneous.

However, commenting the line 103 for "CLIENT.connect()" then produces an unexpected indication that the Dell firmware is not trusted:
FwupdError: firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer (7)

Full example:
[root@localhost firmware_packager]# ./install_dell_bios_exe.py ~/OptiPlex_5060_1.15.0.exe 
Installing to System Firmware
Wrote capsule /tmp/tmpe9t1w4zs/firmware.bin
GUID:      fcd78f90-860a-4dcd-83bd-c088854a01fc
HdrSz:     0x1000
Flags:     0x70000
PayloadSz: 0x1a012f8
Generated CAB file /tmp/tmpe9t1w4zs/firmware.cab
[                                                  ] 0% idle
FwupdError: firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer (7)

This additional situation happens on both Latitude laptops and Optiplex desktops (it's not just that one BIOS shown).

If this second situation needs to be a new/different ticket, please let me know.

hughsie, 2022-01-31:
> then produces an unexpected indication that the Dell firmware is not trusted:

From fwupd's point of view, that's true, no? The signature we're looking for is "distributed by a remote that we trust" and that firmware wasn't signed by any key; it could be the wrong .exe file, or it could be from a malicious source.

pjwelsh, 2022-01-31:
My point should have been that the documentation page (https://github.com/fwupd/firmware-dell) may need to be altered appropriately to give a warning about this conf file required change ("set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer") . I'm not a dev ("set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer") and Dell seems rather sloth-like in updating various firmwares and these are "Urgent" updates so I did it anyway ;)

closed 2022-01-31 by hughsie

hughsie, 2022-01-31:
@dell-client-linux do you want me to make that change? It seems a shame to encourage users to remove the signing requirement just because some vendors are slow to push updates.

CragW, 2022-02-02:
I think the firmware cab is local created, which is not remote signed. In this case, an intentional install should go with root privileged `fwupdtool` rather the client tool.